gnea/grbl #784: Alarm:1 when jogging (after homing)

DavidsCrap, 2020-01-01:
I am running into a problem where I get an ALARM:1 when when I try to jog the machine. I am using UGS and can get as far as clearing the boot alarm and running the homing sequence without issue. Jogging also works in the Setup Wizard, as do the limit switches. If I disable hard limits, jogging works. I assumed it was line noise at first. I tried plugging directly into the arduino with the short, shield cable it came with. I also disconnected the switches after homing. Neither of those changed anything. Console output and pictures of setup below.  Any suggestions would be appreciated. While waiting for any advise, I am going to configure Marlin to see if I get the same issue with different firmware. 

Setup:
Chinese clone (Bulkman 3d) Workbee 1000m 
Arduino mega / grbl mega 1.1f
WINGONEER Prototype Screw / Terminal Block Shield
Stepperonline DM542T drivers
Generic 269oz.in Nema 23 motors from Amazon
Cabling is a mix of shielded cable and the openbuilds Xtension Wire
Xtension Wire limit switches wired just to GND and V+

> Grbl 1.1f ['$' for help]
>>> $$
[MSG:'$H'|'$X' to unlock]
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 770.000    (X-axis maximum travel, millimeters)
$131 = 770.000    (Y-axis maximum travel, millimeters)
$132 = 80.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> $X
[MSG:Caution: Unlocked]
ok
>>> $H
ok
>>> $J=G21G91X100F4000
ok
An unexpected error was detected: (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.
[MSG:Reset to continue]

[image]
[image]
[image]

neilferreri, 2020-01-01:
> Wire limit switches wired just to GND and V+

I'm not sure how you have your switches setup, but this sounds like a short to me. Are you connecting to the limit pin?

DavidsCrap, 2020-01-01:
> > Wire limit switches wired just to GND and V+
> 
> I'm not sure how you have your switches setup, but this sounds like a short to me. Are you connecting to the limit pin?

I had thought the same, but I can't find any. I don't expect that everything would work properly in the UGS wizard if that were the case either. V+ are connected from D10-12 respectively to a terminal block then to the switch. The ground pin is connected to a terminal block, and jumped to a second. The three ground wires connect from that splicer block to the switch block. Pics attached for clarification. 
[image]

[image]

neilferreri, 2020-01-01:
Don't you want the Signal pin connected?
Those switches look to have noise filtering built-in, and need 5v, gnd, and your signal.

DavidsCrap, 2020-01-01:
I haven't found much benefit in the past hooking up the signal pin. I honestly just use them for the easy mounting and connections. Either way, I did just try moving the d10-12 connections to signal and connecting a 5v line from the arduino to each switch. The behavior is still the same. Homing works flawlessly. The switches and jogging work during the setup wizard. After the machine is homed, any further jogging still throws ALARM:1

3RD-Dimension, 2020-01-01:
Is the Pull-Off after it homes sufficant to disengage the switch? Alarm 1 is indicating one of the switches is still activavated.  Try changing setting $27 from 1 something like 1.5 or 2 and see after homing it still throws the error.  Why I say this is because 1mm for pull off to disengage the switch seems a too little, for my limit switches I had to have mine at 2mm for pull-off.

FelisucoVFR, 2020-01-01:
@DavidsCrap do you have pull-up or pull-down resistors?

FelisucoVFR, 2020-01-01:
Not sure how your switches respond with only two wires. Those pcb’s normally have resistors and capacitors on them.
Your connection way I think does not convert them in regular switches.
You can check if voltage (so contact) correctly changes instantly.

FelisucoVFR, 2020-01-01:
Check switch status after homing [M119 at grbl](https://github.com/grbl/grbl/issues/240)

DavidsCrap, 2020-01-01:
After looking with fresh eyes, I'm convinced there is just too much noise in the system. I don't have the right caps to clean it up right now so I'm just going to rewire with an uno and CNC shield. I had wanted to avoid crimping a dozen dupont connectors, but I've already spent 2 days trying to sort this out. I'll revisit once I get some more parts in to clean up the signal and create harnesses to switch between boards with a single plug. Thanks for the suggestions!

closed 2020-01-01 by DavidsCrap

JonnayLin, 2021-07-03:
Hey! Did you ever find a solution to this problem? I'm having the same issue right now!