gnea/grbl #786: How to get actual step count?

amirm, 2020-01-03:
This is not an issue, sorry but I don´t know where to ask - I did not finma a GRBL forum.

As GRBL steps the motors back and forth, does it keep count of how many steps were given? 
Something like a "step position" that increments when the motor steps forward and decrements when the motor steps back?
Can I read it??

I´d like to use this information for precise homing procedure.
Run the homing cycle, and then ask GRBL to move some microns back or forth until step count is a multiple of 64.

Why 64???

Move to a multiple of 16 to stop at start of a microstep cycle.
Move to a multiple of 64 (16*4) to stop at start of a microstep cycle of **step state 0**.

step state??

step state 0 = coil A powered.
step state 1 = coil B powered.
step state 2 = coil A powered, reverse current.
step state 3 = coil B powered, reverse current.

On a 200 step/turn motor, there are 50 stop positions for "step state 0", each one is 7.2° apart.
there are 50 stop positions for "step state 1", each one is 7.2° apart, and 1.8° away from nearest "step state 0".
50 stop positions for "step state 2", 
50 stop positions for "step state 3", 

Thanks

langwadt, 2020-01-03:
sys_position in system.h, don't know if there is a way to read it. Why don't you just reset the stepper drivers after homing? that'll also guarantee that what you think is "step state 0" is really "step state 0"

amirm, 2020-01-03:
Why don´t I reset the stepper drivers... because on most boards the reset signal is "global", and it resets all the drivers and GRBL. I´d need to cut & wire the board to isolate the reset lines - and then have 3 output signals.., and control them! Software is far easier to modify.

I dug a little, and changed **report.c** :

```
void report_realtime_status()
{
  ...
  int32_t current_position[N_AXIS]; // Copy current state of the system position variable
  memcpy(current_position,sys.position,sizeof(sys.position));
  ...
  #ifdef REPORT_CONTROL_PIN_STATE 
    printPgmString(PSTR(",Ctl:"));
    print_uint8_base2(CONTROL_PIN & CONTROL_MASK);
  #endif

  // --- MY CODE START ------------
  printPgmString(PSTR(",STEP_COUNT:")); 
  for (idx=0; idx< N_AXIS; idx++) {
    print_uint32_base10(current_position[idx]);
    if (idx < (N_AXIS-1)) { printPgmString(PSTR(",")); }
  }
  // --- MY CODE END ------------
  
  printPgmString(PSTR(">\r\n"));
}
```

I´ll report as soon as I have results. May take a while...

Thanks for pointing the way, langwadt!!

FelisucoVFR, 2020-01-04:
Sorry @amirm but I cannot see any advantage on that homing. I am probably missing something. There is no more accuracy nor torque on those positions, but even if there were both, it is homing, your improvements are needed while working.
Sure internally grbl is counting steps as it is the way it works. Follow were steps/mm is used.

amirm, 2020-01-06:
Maybe you are right! Maybe I´m chasing ghosts...

My concern is the Z axis of a laser machine, where I need very precise and repeatable homing to always get the same focal point.

Like I said, there are some tests to be done, and I´ll post the results. So far, the modification I did has worked - I can read step counts.
More important than that, it gave me confidence to change the code and compile it to suit my needs. I´ll share my findings.

wlaw, 2020-01-06:
In this case I would use a Proximity switch to home the Z-Axis.
The prox switch should provide a very accurate feedback and allow you to
positions the laser very accurately moving down.

using just a micro switch is close but not as accurate.

doppelhub, 2020-01-24:
I report step count as follows (example):
`count=sys_position[X_AXIS];
printInteger(count);`