gnea/grbl #788: NC Limit switches not working properly

arminlinder, 2020-01-03:
After spending a whole day trying to resolve this issue, I dare to bug you. I set up GRBL on an Arduino Uno to run a 3 axis CNC mill. The steppers run fine, G0 commands work, now I added end-stop switches (NC type, wired to GND).

Since then the machine will always throw an Alarm:1 when any Stepper starts.

My setup is:

$21=1
$5=1

everything else at the defaults. See the end of this post for all config values.

If I enter any motion command, e.g. G0 X -1, the motor does a faint "click" and stops immedately, and grbl throws "ALARM 1". It obviously thinks, it has hit and end-stop, though there isn't one. This happens in the same manner every time (tried at least 50 times now ...). Here's a log of such a "session":

ok
?<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
G0 X 10
ok
ALARM:1
[MSG:Reset to continue]
Grbl 1.1h ['$' for help]
[MSG:'$H'|'$X' to unlock]
?<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>

You see, the machine didn't even move a step.

Already tried/ruled out:
- reset the error, machine ready, waited a while, nothing happened, then pressed any one of the end stops by hand: "ALARM:1" appears at once, like expected. So I guess that the switches are wired correctly. 
- cabeling. As I said, I ended up disconnecting all end-stops, and connecting all three endstop-pins to GND via jumper cables, and get the same Alert as soon as I enter any motion command. I think this rules out all possible cabeling errors.
- defective Arduino Uno. I write myself a little sketch which enables pins 9,10,12 with INPUT_PULLUP, and I digitalRead the correct states: LOW if the NC switch is not pressed, HIGH if it is pressed.
- just for fun, I added noise suppression and debouncing via 470 Ohms to +5V and 10nF to GND like shown in the docs, problem persists.
- measured the voltage at pin 9,10 and 12 using an oscilloscope, all three are at 0V, and jump to 5V if I press any switch. No spikes noticeable.
- finally I loaded the very same hardware with a crude "Bit-Banging" stepper control routine which uses digitalRead after every Step to check whether a switch has triggered. The routine is meant to "torture" my setup, running the stepper at full speed approx 10cm in one direction, and back. I let it run for several hours, and got not a single end-stop alert. Not one. Just to check, I pressed an end-stop manually, and the stepper hatled at once, so the switches are working.  
- I also disconnected the stepper, and let the Arduino send the signals into hyperspace, the error came immedately. If it is any kind of interference, maybe with the ENA signal, it's really a nasty bugger.

AT this point I thought I ruled out any hardware issues, which shifts attention to a configuration issue. But I followed your docs, and double-and triple checked everything, and found no error, nothing smelling like it could make GRBL something like hyper-sensitive when it comes to checking end-stops. In parallel, my oscilloscope was running all the time, and I never noticed any glitch in the signals.

Guys, I am stuck, and any help would be greatly appreciated.
For completeness, here is my complete config:
ok
$0=10
$1=25
$2=0
$3=0
$4=0
$5=1
$6=0
$10=17
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=80.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=5.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
ok

Any ideas?

DenisDeni, 2020-01-04:
$5=0

arminlinder, 2020-01-04:
$5=1 should be correct for a NC switch wired to GND. From the Docs:

"By default, the limit pins are held normally-high with the Arduino's internal pull-up resistor. When a limit pin is low, Grbl interprets this as triggered. For the opposite behavior, just invert the limit pins by typing $5=1."

IMHO, $5=1 is correct for my seetup.

Btw, I tried NO switches and $5=0 as well, and got the same behaviour.

PicEngraver, 2020-01-05:
My suggestion is to save yourself much frustration and headaches, as well as lost work, disable the hard limits ($21=0), and use the switches for homing only ($22=1).  For limits, use soft limits ($20=1).  Works great.

arminlinder, 2020-01-05:
IMHO, both need to work. Currently I am trying to decouple the Arduino inputs with Opto-Isolators, and rework the shielding of all end-stop cables. I guess this will very likely fix the issue. Seems I under-estimated the sensitiveness of Arduino inputs to electrical noise, seems to be much more of a problem if a pin is used in interrupt mode.

PicEngraver, 2020-01-05:
I agree it is not a grbl issue; it is likely a hardware issue caused by EMI and static voltage buildup.  Here's hoping you can come up with a solution that fits all, but my cynical self has doubts.  Seems so much easier and reliable to just use soft limits.   It does become more problematic in low humidity environments in my experiences.  A sincere Good Luck.

arminlinder, 2020-01-08:
Resolved.

The problem hadn't anything to do with GRBL, it was caused by a faulty cable in the wiring.

Thanks for trying to help,

Armin.

closed 2020-01-08 by arminlinder

palin, 2021-11-26:
I had exactly the same issue as described by the owner of this topic - arminlinder. I wasn't able to move any axis because of immediate "hard limit" alarm and stop even though the limit switches were not physically triggered.

In my case the issue was caused by **cables interference**. Here's how I debugged and proved that issue:

- I unplugged all cables that were connected to limit switches and to my arduino (so the arduino limit switch ports 9, 10, 12 became empty)
- I set $5=0 (invert limit pins = 0)
- I set $21=1 (hard limits enabled), $22=1 (homing cycle)

With these settings I was able to move all axis without any issue and no alarm has been triggered. 

At that moment I connected 2 new cables - first one to the limit switch port (9, 10 or 12 - doesn't matter) and the other one to the GND - I was still able to move any axis after that.
The last step was to physically move the cables (that I just connected) next to the motor cables - just putting them on top or very close aside so they touched each other. 
When I tried to move any axis the hard limit alarm got triggered immediately! I moved the cables about 20cm aside of the motor cables and retried. This time axis were moving fine again! 
I retried this process a couple of times to make sure it's that issue. **The solution would be getting good shielded cables!**

I hope that it will help someone because it was driving me crazy!