gnea/grbl #792: Hello everyone, I wanted to create an M12345 command by giving an example that when called, go to a specific position.

soinel, 2020-01-10:
Hello everyone, I wanted to create an M12345 command by giving an example that when called, go to a specific position.

I see that it could be created in gcode.c and the format would look like this:

case 12345: HERE IS WHERE I DON'T KNOW WHERE TO GET IT; break;

As I do? for what .....
 
       case 12345: G0 Z50 F1000; G0 X150 Y10 F1000; M0; break;

would this be correct ??? No right .... can someone explain to me how the correct way would be?

thanks

MARIOBASZ, 2020-01-18:
You could, but I would do it through a macro or an order in the program that controls grbl.
Personally, I use bCNC, there it is simple to assign a button to execute the movement (s) you order

doppelhub, 2020-01-24:
As MARIOBASZ mentioned, probably best to do this in the g-code itself (e.g. G90 G21 G56 X-39 Y-10 Z-4)... but if you want to build it into an M command, you'd add your case number to case 'M' in gc_execute_line(line).

MARIOBASZ, 2020-01-24:
You can add an M 1234 command also to grbl to do something edoecifico, you have to modify some / some grbl files (I must look to see how I did it, I will need this shortly for an ATC design project for my cnc )

soinel, 2020-01-24:
doppelhub thanks,
Could you give me an example?

soinel, 2020-01-24:
case 12345: gc_execute_line(G0 Z50 F1000; G0 X150 Y10 F1000; M0;); break;

it would be like this?

doppelhub, 2020-01-24:
In theory, yes.  The line would need to be properly formatted (no spaces, no comments, all caps):
`G0Z50F1000`, `G0X150Y10F1000`, etc.

Note that you'll need to store these lines somewhere besides memory (e.g. using 'PSTR("G0X150");', which forces the compiler to dump the string to program space.  Better yet, I'd store this string in EEPROM, which will allow you to change the position without recompiling.  grbl's EEPROM map has plenty of room.  I recommend structuring your code similarly to $N (startup lines).  Take a look at the EEPROM memory space (settings.h) to figure out where you want to store it.  I've added several values in my grbl variant (https://github.com/Defdist/grbl1v1g_GG3_GrBLDC3v0)... diff my settings.h with grbl master and you should get a good understanding on what's open.  I also created a graphical EEPROM map in openoffice calc here (https://github.com/Defdist/grbl1v1g_GG3_GrBLDC3v0/blob/master/doc/architecture/grbl%20EEPROM%20space.ods).

soinel, 2020-01-24:
perfect, thank you very much.

I'll get to see if it works for me.

soinel, 2020-01-26:
case 1234: gc_execute_line(G0Z50F1000 , G0X150Y10F1000 , M0 , G0X0Y0F100 , G0Z5F100 , G10P0L20Z10 , G38.2Z0F100 , G10P0L20Z0 , G0Z40F1000); break;

no found

exit status 1
'G0Z50F1000' undeclared (first use in this function)

soinel, 2020-01-26:
if I run a single line if it works

gc_execute_line("G0Z40F1000"); break;

soinel, 2020-01-27:
I tried to do it separately but it does nothing, I don't get it to execute all the lines. Can you guide me?

case 1234: gc_execute_line(G0Z50F1000);
gc_execute_line(G0X150Y10F1000); 
gc_execute_line(M0);
gc_execute_line(G0X0Y0F100);
gc_execute_line(G0Z5F100);
break;

soinel, 2020-01-28:
Hi everybody how are you.
I tried to do the following, but it doesn't work either.
Any ideas or suggestions?

	  case 1234: ;
		int contador = 0; // Inicialización del contador

		while (contador<10) {          // Condición
				  if (contador == 1) { 
					gc_execute_line("G0Z50F1000");
  					}
				  if (contador == 2) { 
					gc_execute_line("G0X150Y10F1000");
  					}
				  if (contador == 3) { 
					gc_execute_line("M0");
  					}
				  if (contador == 4) { 
					gc_execute_line("G0X0Y0F100");
  					}
				  if (contador == 5) { 
					gc_execute_line("G0Z5F100");
  					}
				  if (contador == 6) { 
					gc_execute_line("G10P0L20Z10");
  					}
				  if (contador == 7) { 
					gc_execute_line("G38.2Z0F100");
  					}
				  if (contador == 8) { 
					gc_execute_line("G10P0L20Z0");
  					}
				  if (contador == 9) { 
					gc_execute_line("G0Z40F1000");
  					}
				  if (contador == 10) { 
					contador = 0;
					break;
  					}
  			contador++;                  // Contador se incrementa
			}

	break;

soinel, 2020-01-31:
Hello everyone, I think I have found the problem, the problem generates some instructions, when commenting on the problematic instructions the code continues, in the case of M0 which is a pause I write what is internally M0 and it works but the following instruction prevails and Skip the break.

	  case 1234: 
                 gc_execute_line("G0Z50F1000"); 
		 gc_execute_line("G0X150Y10F1000");
		 //gc_execute_line("M0");
		 gc_block.modal.program_flow = PROGRAM_FLOW_PAUSED;
		 gc_execute_line("G0X0Y0F1000");
		 gc_execute_line("G0Z5F1000");
		 //gc_execute_line("G10P0L20Z10");
		 //gc_execute_line("G38.2Z0F100");
		 //gc_execute_line("G10P0L20Z0");
		 gc_execute_line("G0Z40F1000");
		 break;