gnea/grbl #795: Can automatic reporting be turned off?

arminlinder, 2020-01-13:
I'd like a status report when and only when I send a "?". No messages, no alarms, no nothing. Can this be achieved through configuration? Thnx, Armin.

zeevy, 2020-01-13:
There is no automated report, 
grbl sends status only after receiving "?"

arminlinder, 2020-01-13:
"No messages, no alarms, no nothing. "

:-)

arminlinder, 2020-01-13:
And btw, why doesn't "?" work through a homing cycle?

zeevy, 2020-01-13:
"?" does not work during homing, because grbl don't know the position of the machine for sure.

arminlinder, 2020-01-13:
I see. 

Thanks for the clarification!

Armin.

closed 2020-01-13 by arminlinder

langwadt, 2020-01-13:
I think "?" during homing has been discussed before, afair it was a simple oversight that it isn't there, but it was left as is in case someone expected it

arminlinder, 2020-01-14:
If so, I strongly encourage that the problem is fixed. I'd rather see false coordinates on my computer screen while homing than nothing at all, so in case something mechanical goes very, very wrong I can see that GRBL has tried. After all, if false corrdinates bother me, I can always ignore the corrdinates during homing, but use their (relative) change to display a kind of progress indicator.