gnea/grbl #796: How to ignore Commands

iplayfast, 2020-01-18:
I've got a project that is a sand table. The idea is a ball bearing is dragged through sand on a table top.  This works fine, however I want to send it 3d printer gcode files, where there are move relative commands or other commands that grbl doesn't know. It get's stopped in it's tracks.  What I would like is an option that for home, and move commands all other commands are ignored. (heater, z movement, etc). 

Do you think this is possible?

zeevy, 2020-01-19:
If those commands does not produce alarm or crtical errors on grbl side then those can be ignored on the gcode sender gui you are using

doppelhub, 2020-01-24:
I don't completely understand what you're describing, but unknown grbl commands can be formatted/removed prior to sending them to grbl (i.e. on the host layer).  The simplest method is to format non-g-code commands into comments '(' and ')'.

iplayfast, 2020-02-10:
Yes I know that is one option. The request was to ignore unknown commands in the grbl firmware. I'm getting critical errors, when sending 3d printer gcode to the grbl. I think it has to do with temperature settings.  Which now that I think of it, ignoring wouldn't help, because the code sender waits until the proper temp has been reached. I think this issue should be closed. Sorry for wasting your bandwidth.