gnea/grbl #798: Not homing

Brucejb, 2020-01-20:
So I’m having trouble with my homing, I have all 6 limit switch’s connected and work they are set to NO= Normal On. Now something I’m not sure that might make a difference in how everything works is that the Z and Y axis are connected I guess would be considered in series, as the second limit switch is jumping off the first one, now the X axis both of the limit switch’s go straight to the board, they don’t connect to each other, the reason why I did this is that my GRBL Board has 6 sets of pins for each 2 sets for each axis but sense I had to make some of my own pin connections I was running out of room, so my question to that is do I need to have them all in series or does it matter if you have two in series and the other axis go straight to the board rather then being connected in series, again if I’m saying it wrong correct me please I’m very very new to this so please don’t talk fancy, my vocabulary in this field is very very limited I’ll post a video of my machine and problem https://youtu.be/uzN3Ts-BpgA here’s 2 pictures of my board and IMPORTANT TO NOTE : this is a GRBL BOARD Brand woodpeker CnC camxtool, not a Arduino or Arduino and GRBL combined
[image]
[image]
Thanks for your time and support 😊

Brucejb, 2020-01-21:
Other thing I failed to mention is that the version of GRBL I’m using is GRBL 1.1

terjeio, 2020-01-21:
NO is "normally open", not "normally on". If you put switches in series then you have to use "normally closed" (NC) switches to be able to detect a status change if only one is actuated. Also be aware that per default grbl is configured for normally open limit switches.

PicEngraver, 2020-01-21:
If your board is ver.3.4, the X and Z limit inputs are reverse marked on the board.

Brucejb, 2020-01-21:
Thanks lol that’s what it was 😂

closed 2020-01-21 by Brucejb