gnea/grbl #799: No "?" status feedback while homing ($H)

arminlinder, 2020-01-21:
Hi folks,

I already have been told that there probably were reasons why the $H command won't allow for any status reports until done. Since nevertheless having it would be handy for my ongoing project, I am upon to add this functionality to the GRBL source myself, I think I can, with some efforts, find out where I need to make changes.

But since I haven't too much time to spend ...

(1) Have there been other reasons disable the ? report in mc_homing_cycle than the fact that the corrdinate display may be meaningless? Timing issues for instance? If returning stats infos would have adverse effects on the homing process, e.g. the accuracy, please tell me, so I dump this idea.

(2) Anyone deeper in the code which has already done this? Would be a great time-saver if there is something I can copy&paste. 

Thnx, Armin.

terjeio, 2020-01-21:
1. I believe timing issues, generating a ? response may take a long time to complete impacting the accuracy (switch state is polled). A comment in the source (limits.c) hints at that.

2. I do not know of any.  A simple test will be to check for the report flag and generate the report in the homing do...while loop and see if that causes any problems. If only accuracy is then a problem change to interrupt handling of limit switches state, preferably without software debounce.

doppelhub, 2020-01-24:
'?' isn't reported while homing because grbl is running at a much lower level than normal.  While homing, grbl disables limit switch interrupts, bypasses mc_line(), and directly plans motion to move either towards or away from each limit switch.  grbl then directly executes motion and then rapidly polls the limit switches until they trip.  As soon as a limit switch trips, the steppers are rapidly stopped (even though the planner briefly thinks it's still moving).  The only way to end the low-level homing routine is to trip the limit switch, travel too far (without tripping limit switch), fail to pull-off a limit switch, or send cycle_stop/reset commands... everything else is ignored.  

There's really no reason to send '?' during homing... nothing useful is returned while homing, plus handling that interrupt would affect limit switch repeatability.

arminlinder, 2020-01-24:
Thank you for your infos, they're really useful, and I think I'll do exactly what you suggest. You are right guessing that all I am after is getting some feedback for m custom GUI while the home cycle is running to keep the GUI alive. After all it takes approx. 2 mins to home my longest spindle (1.2m) spindle, if the toolhead happened to stop near the far end, and I feel there shoul be some activity on screen.

Thnx again!

doppelhub, 2020-01-24:
Does your machine normally take more than 2 minutes to travel 1.2 meters?  That is really slow.  There are two parameters you can set while homing... 'seek' can be really fast... after that, the (slower) homing feedrate takes over, but it only needs to move a short distance.

arminlinder, 2020-01-24:
While the machine is set to run a may speed of approx 1200m/min, I left homing at the default, which is, I think, 300 m/min for phase I and even slower for phase II.

I think this is a good setting for my machine. All G-Code programs which come to a regular end bring the toolhead back to 0/0/0, which is right near the end switches. Usually homing takes just a few seconds. If the machine has to travel the whole length of the spindle something has gone horribly wrong, maybe a power outage, or someone hit the emergency stop for some reason. 

I do initiate a deliberately slow homing procedure so the one who eventually removes any obstacles from the toolpath and then disarmes the emergency stop and tells the machine to reset has plenty of time to hit the emergency stop again if he needs to.

arminlinder, 2020-01-25:
I think I have sufficient infos to make the required changes, thanks to doppelhub.

If someone ever thinks about leaving the feedback alive until homing (Phase I) has finished - I'd encourage this. It IMHO won't do any harm, and makes a GUI more responsive during homing.

closed 2020-01-25 by arminlinder

doppelhub, 2020-01-25:
I agree it won't do any harm to enable realtime status reporting during initial homing pass.  This is probably not more than 4 lines of code to implement.

tomolenet, 2020-12-23:
Because `?` is disabled while homing there is no possibility to display the `x`, `y` and `z` coordinates while homing?