gnea/grbl #806: Wiring optical endstops

draregerreip, 2020-02-08:
Hi,

I read posts 96, 171, 380 and 594 as well as the wiki on wiring mechanical endstops, however it didn't help me I tried to wire the optical endstops on my laser engraver. At the moment, the electronic is based on arduino uno + cnc shield v3 with grbl 1.1. 

Building this machine, I chose to use optical endstops  which have a 3 pin connector (Ground/Signal/5V). When I have one endstop connected (S1 or S2 on the picture below), il works fine.

[image]

However, when I connect the two endstops (S1 and S2) it doesn't function at all. Reading the wiring mechanical endstops wiki, I understand how it can work with mechanical endstops but I can't see how to apply it to my specific problem.

I believe I could make it work with a logical gate based on a an [74LS32 ](https://components101.com/ics/74ls32-or-gate-datasheet-pinout-and-specs) IC  but it looks like using a canon to swat a fly ...

Would anyone have a simpler idea on how to make it work ?

Thanks in advance for your suggestions

langwadt, 2020-02-08:
which optical end stop are you using?

draregerreip, 2020-02-09:
I am using reprap ones (originally for a 3D printer) which I had in stock, bought them on aliexpress a while ago [https://www.aliexpress.com/item/32434841034.html?spm=a2g0s.9042311.0.0.27424c4dvAAF6C](https://www.aliexpress.com/item/32434841034.html?spm=a2g0s.9042311.0.0.27424c4dvAAF6C)
They function well, are cheap, with no interference that I can detect and good repeatability. I like them better than the mechanical ones.
[image]

The signal goes from below 0.5V when nothing bloks the light to 4.8V as soon as the light is blocked (maybe there are intermediate values but I have no way to measure them).

langwadt, 2020-02-09:
what in your parts bin? two diodes or two npn transistors ought to do it

draregerreip, 2020-02-09:
well my part bin is very very poor, i've got enough resistors and capacitors but the only transistors I have are a few (5) 2N2222 NPN ones (package looks like TO92 for what it's worth).
I have some theoretical knowledge, but no real practical experience, so if you can draw me a quick diagram I would be grateful.

langwadt, 2020-02-09:
something like this, 

[image]

draregerreip, 2020-02-12:
Thanks a lot. It took me a while but I finally got it to function properly using your drawings.
For the people who would read this post in the future, here is what I came up with from langwadt drawing.
[image]