gnea/grbl #807: How detect a command has been actually carried out?

arminlinder, 2020-02-09:
Hi all,
I am working on integrating a GRBL controlled CNC axis into an application. Generally the serial communication works well, but I haven't found a satisfactory solution to the following problem:

*    Most GRBL commands, like G0X... will be acknowledged by "ok" when received/buffered, but when actually carried out and finished. if I intend to send a sequence like this:

$110=300
G0X400
$110=80
G0X0

and send the next command after I have received "ok", the second set speed ($110=80) command will end up in error 8, since I may change speed only if the machine has finished processing the previous G0 command.

I tried several approaches like to wait for "idle" and "run" state by polling the machine state using "?" commands, but the results were unsatisfactory.

Can anyone advise me how I can organize my communication so I can reliably detect when GRBL is ready to receive the next command?

Thnx, Armin

langwadt, 2020-02-09:
don't do that. $110 is something you set once to the limits of your machine, it is stored in eeprom.
G0 is for moving as fast as the machine can, if you want to move a specific speed use G1 instead that is what it is for

G1X400F300 
G1X0F80

arminlinder, 2020-02-09:
Interesting! I didn't know this. This will indeed resolve some if not all of my issues.

I do still see a problem to detect when the very last command of a G-Code program has been carried out. If I end my G-Code with G1X0F1000 I get 'ok', but the machine will still be on its way to 0/0/0 for several seconds.

In my former attempts I learned to fear the approach to simply poll for "idle".

Any better approaches?

Thnx, Armin.

langwadt, 2020-02-09:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#synchronization

arminlinder, 2020-02-09:
Even better :-)

Thanks!

closed 2020-02-09 by arminlinder