gnea/grbl #810: After stop (!) and soft reset (0x18) controller positions were mismatched

luben111, 2020-02-11:
I noticed a strange behaviour when stopping the machine with (!) and resetting it with 0x18 - if I send new movement commands the real position is displaced by some offset (about 10mm on X, Y and Z). This happens only once (after the first ! and 0x18) and repeating the procedure is not changing the head position.
It's strange that the machine is reporting correct XYZ positions through (?), just the head moves away from the correct positions.

My guess is that the soft reset clears some information inside the controller and even if the working position is reported the same after movement command the head moves with offsets.

One question - when stopping the movement with (!) is there another way to go away from HOLD state except the soft reset (0x18)?

luben111, 2020-02-12:
It seems that the root of the problem is that machine coordinates change during soft reset where working coordinates remain the same. If your GRBL setup uses homing and non zero value of Pull_off_distance after the homing was completed the working and machine coordinates will differ with the homing Pull_off_distance.  Reading the working coordinates before and after soft reset will not show any difference, they look absolutely the same, but machine coordinate changed. Like result when you start a new movement after the soft reset you'll witness that head jumps to offset positions.

After some investigation I found the right sequence of events and circumstances which is causing the problem:
1. Homing enabled
2. Pull off distance ($27) is not zero
3. Homing cycle was completed successfully
4. Soft reset (0x18) was sent to the machine
5. After soft reset was completed the new movements will go to offset (X, Y and Z) by Pull_off_distance positions.

PicEngraver, 2020-02-12:
Maybe this will help others experiencing unexpected loss of zero positions.
https://github.com/grbl/grbl/issues/239