gnea/grbl #813: inverse motion

Rigo5507, 2020-02-15:
Hi
I bought my CNC for about a year ago and once it was assembled I got this crazy behaviour:
Whn I press for right it goes left , If I press forward it goes backwards. and the spindle goes opposite  ryin to engrave in the air while lifting the blade it goes down into the table. So totally oposite of what I expected.

I ve been using Grbl 0,81 and upgraded to 1.1...still the same problem.  I remember I did change the CH341 to CH340  (mybe the other way around)  and then it seemed to at least get the axis to behave correct ...but I'm not sure.

I Also tried to use grbl controller 3.61 and I needed to set th x,y axis in inverse mode but still got problem with the Z ( spindle up/ down movement).
Allcables was already assmbled and Im using the Developower 2.0 ( the one with 12 and 24 V adapter)

So anyone knowing the Issue I've got...I really need help 

Richard

lalo-uy, 2020-02-15:
Look like all your motors ara reverse wired. Try to invert at least one and
check.

langwadt, 2020-02-15:
change https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask
or swap the leads for one of the coils on each motor

Rigo5507, 2020-02-15:
Thanks ! I changed the leads...for some reason the disributer has assemled the connector to cables 180 degrees wrong. Iswaped them and that worked. Now I got the some other problem...the size of the images does become very very tiny...where do I adjust the the proportion and scale? Inside the code or is there some settings..?

Once again...thanks for the help with the inverse problem.

langwadt, 2020-02-15:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm