gnea/grbl #815: Grbl hangs on rapid after job complete

hairu, 2020-02-18:
After running the program in the error.txt attachment and then issuing a G90G0X0Y0Z0 via MDI, Grbl either stops responding/hangs in the run state or behaves erratically such as moving to a random location.  I am able to reproduce the issue in the official hex of 1.1h, and the Mega2560 port.  I've reproduced it in my own sender program (https://hackaday.io/project/169898-touchscreen-control-panel-for-grbl), UGS, and bCNC.  I suspect that the issue is somehow related to my settings file, so I've included it as well.

[settings.txt](https://github.com/gnea/grbl/files/4217082/settings.txt)
[error.txt](https://github.com/gnea/grbl/files/4217076/error.txt)

Thanks

zeevy, 2020-02-18:
It run successfully, i have not faced any issue on 1.1h

hairu, 2020-02-18:
I don't have any preprocessor configured that I know of.  Did you issue the G90G0X0Y0Z0 afterwards?

langwadt, 2020-02-18:
250steps/mm * 10000mm/min /60 = 41.6kHz, that is well above the recommended ~30kHz maximum

hairu, 2020-02-18:
Reducing the steps/mm to 40 solved it!  I was running off of arduinos disconnected from my CNC for software development purposes and didn't notice that the settings were off.  Thank you for your help.

closed 2020-02-18 by hairu