gnea/grbl #819: Reporting of immediate Spindle/Feed rate override value changes

Slixxor, 2020-02-24:
Hi,

I've been designing a tool for GRBL to adjust the immediate overrides for spindle and feed rate. 

I'm sending the respective commands e.g: "0x9B" and  not seeing any changes to "FS:" reporting on the "?" command.

Can I confirm the Spindle speed and Feed rate values actually change the values or is it just assumed? 

It looks like my commands are simply being ignored but i'm not sure. Is there a setting in firmware I need to change to allow overrides?

AlexHolden, 2020-02-25:
I implemented the overrides in my sender and I can confirm that it worked as expected. The reported values do change. I didn't need to alter any grbl settings to make it work.

Slixxor, 2020-02-25:
> 
> 
> I implemented the overrides in my sender and I can confirm that it worked as expected. The reported values do change. I didn't need to alter any grbl settings to make it work.

I'm using .NET framework for the software. so for example, i'm sending Chr(154) for Spindle -10%

Works fine when sending "Chr(24)" for E-stop. I get OK responses to Chr(154) but I don't see the value change in "?" response.

Are you in a particular state when it's happening? I.e Jog.

I've tried doing a long slow jog and sending the data but it does not update. Have tested on a few chips. See data log output here:

<< ?
>> <Run|MPos:0.000,353.575,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,353.055,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,352.535,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,352.010,-50.000|FS:100,1000|Pn:XYZ|Ov:100,100,100|A:S>

<< ?
>> <Run|MPos:0.000,351.700,-50.000|FS:100,1000|Pn:XYZ>
ok

<< ‘
>> <Run|MPos:0.000,351.490,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,350.970,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,350.455,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,349.935,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,349.415,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,348.890,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> <Run|MPos:0.000,348.375,-50.000|FS:100,1000|Pn:XYZ|WCO:0.000,347.415,-50.000>

<< ?
>> <Run|MPos:0.000,347.850,-50.000|FS:100,1000|Pn:XYZ>

<< ?
>> ok

<< G91 G21 G01 Y1000 F100
>> <Idle|MPos:0.000,347.415,-50.000|FS:0,1000|Pn:XYZ|Ov:100,100,100|A:S>

AlexHolden, 2020-02-26:
The feed rate override commands have an effect on the axis movement speed while performing a G1/G2/G3 move. The rapid rate override does the same thing for G0 moves. Neither of them have any effect on the speed of realtime jogging moves.

Spindle speed and spindle speed overrides aren't affected by jogging. I just attempted to change the spindle speed override during a long jog with the spindle turned on (though I'm not sure why you would want to do that), and it worked as expected, immediately changing the reported spindle speed without interrupting the jog move.

AlexHolden, 2020-02-26:
By the way, the overrides are all limited by whatever you have set as the maximum values in the grbl configuration. For example you can't use the spindle speed realtime override to increase past the $30 maximum spindle speed value.

Slixxor, 2020-02-26:
Ok, Issue found. I'm sending ASCII data to the SP. I've changed it to a byte array and the issue appears resolved.

closed 2020-02-26 by Slixxor

MeJasonT, 2020-02-27:
@Slixxor 
It looks like you are creating a user interface is this correct.  I would be most interested if you are.

MARC07150, 2020-03-03:
Hello.
I am testing Arduino1.8.5 to GRBL 1.1.g (Arduino NANO), I use 2 motors, by the IDE I put the code "G01 F180 X1600 Y800".
To stop the engines I code "!", To restart "~". I would like to use the real-time command (Overriding: Feed rate)
The commands are:
     0x90: Set 100% of programmed rate.
     0x91: Increase 10%
     0x92: Decrease 10%
     0x93: Increase 1%
     0x94: Decrease 1%
But what coding I have to write in the Arduino IDE for these commands to be active.
Cordially.
Mark.

Slixxor, 2020-03-04:
> Hello.
> I am testing Arduino1.8.5 to GRBL 1.1.g (Arduino NANO), I use 2 motors, by the IDE I put the code "G01 F180 X1600 Y800".
> To stop the engines I code "!", To restart "~". I would like to use [...]

Hi Marco, these are enabled by default. No extra work is required. What language are writing your control software in?

Slixxor, 2020-03-04:
> 
> 
> @Slixxor
> It looks like you are creating a user interface is this correct. I would be most interested if you are.

Yes indeed :)

https://www.youtube.com/watch?v=eANuoBiRJTM&t=2s

MARC07150, 2020-03-04:
Good evening thanks.
I use the IDE of ARDUIINO in French for the tests, but I would like to make the commands in interface PYTHON.
Marc.

Slixxor, 2020-03-04:
> 
> 
> Good evening thanks.
> I use the IDE of ARDUIINO in French for the tests, but I would like to make the commands in interface PYTHON.
> Marc.

Ok, well the issue I had was the serial port itself on the PC. When you send ASCII chars they are limits upto character 127. So I had to convert it to a byte array. Here is a snippet of the send routine that works for the higher values.

` Private Sub SPSend(ByVal Dataval As String, Optional ByVal Async As Boolean = False)
  If Dataval.Length = 1 Then
   Dim Strval As Byte() = System.Text.Encoding.GetEncoding(1252).GetBytes(Dataval)
   SPN.BaseStream.Write(Strval, 0, Strval.Length)
  Else
   SPN.Write(Dataval)
  End If

 End Sub`

MARC07150, 2020-03-04:
OK thank you, I test and I tell you.
Marc.