gnea/grbl #822: Details about communication with Grbl 1.1

wnelis, 2020-03-01:
Being new to Grbl, I am trying to understand de details of the communication with Grbl. Those details are needed as the plan is to write my own driver (non-GUI probably). Grbl v1.1 is in my case implemented on an Arduino Nano.
Below is a list of specifications for the communication as I understand it now, partly documented on the github site of Grbl, partly found by googling and partly just my best guess.

Any comments with respect to correctness and completeness are appreciated.

- The communication uses asynchronous serial transfer, 115200 Baud, no parity, 8 data bits and 1 stop bit (115200N81). It is not possible to set another speed.

- The communication is full duplex.

- The communication uses an extended form of ASCII, in which each symbol can be represented in one byte (octet).

- The basic unit of communication is a line. A line is delimited by end-of-line symbols, which are ASCII symbol CarriageReturn when sending to Grbl, and ASCII symbols CarriageReturn plus LineFeed when receiving from Grbl. The end-of-line symbols are not considered to be part of the line.

- The data stream to Grbl consists of a mix of three types of messages, the g-code blocks, the system commands and the real-time control commands.

- A g-code block consists of zero or more g-codes. The g-codes are documented at various places. The precise syntax and meaning is outside the scope of this list.

- An empty line is considered to be a g-code block with zero g-codes in it. As a result, the response on sending an empty line is always 'ok', and thus never an error message.

- Each g-code block results in exactly one response, either 'ok' or an error message. The regular expression to describe (recognize) an error message is '^error:\d+$' (the single quotes are not part of the regular expression). The response is thus recognizable with regular expression '^(?:ok|error:\d+)$'.

- The maximum length of a line is 80 symbols, not counting the comment sections in a g-code block.

- The system commands all have a '$'-sign at the start of the line.

- One system command is contained in one line, while one line contains at most one system command.

- A real-time control command consists of a single symbol, thus it is not delimited by end-of-line symbols.

- A real-time control command can be sent at (almost) any time. It can not be sent if another real-time control command is being handled by Grbl or when a (soft) reset is in progress.

- Only real-time control commands '^X' (0x18, soft reset) and '?' (query status) give a response. The first one results in the welcome message, the second one in a line delimited by chevrons. The response of a status query is thus recognizable with regular expression '^<.+>$'.

- The other real-time control commands do not give a response. Still if multiple commands are sent, the second and later commands will (might) be ignored until it has executed the first received command. Thus for flow-control purposes, the time between two successive real-time commands without response should be at least 25 [ms].

- All the responses should be identified, to detect the push messages. These are not the (direct) result of a query, but are pushed by Grbl. The push messages are not a part of the streaming protocol (flow control method).

- There are two flow control methods, called send-response (fc-sr) and character-counting (fc-cc).

- Both methods only applies to the streaming protocol, thus only to g-code blocks and system commands.

- In case of fc-sr, a g-code block or a system command is not sent until the response on the preceding line (with either a g-code block of a system command) is received. Thus it is a flow control method with a transmit window size of one (line).

- In case of fc-cc, a line, with either a g-code block or system command, is sent if there is enough space in the receive buffer of Grbl to accommodate the complete line, including the end-of-line symbol. Upon receipt of a response, the buffer space used by the line leading to the response is free to use for a next line.

- Fc-cc can not be used if the line causes the EEPROM to be written. Prior to sending the line, the sender should switch to fc-sr. Upon receipt of the response of the line, indicating that the EEPROM modification is completed, the sender can switch back to fc-cc.

- A command will (might) modify EEPROM if it matches one of the following regular expressions: '^\$RST=.*$', '^G10 L20? P\d+$', '^G28\.1', '^G30\.1', '^\$\d+=.+$', '^\$I' and '^\$N[01]=.+$'.

- While a reset is going on, nothing can be sent to Grbl. Once the welcome message is received, lines can be sent to Grbl.

- Upon sending the system command '$C' which terminates the check mode, communication must be halted until the welcome message is received.

- If synchronization between sender en Grbl is lost, for instance because an expected response is never sent, there is no easy way to regain synchronization.

Apologies for the long post. I really like to know if I am on the right track.

MeJasonT, 2020-03-02:
most of what i have read agrees with my understanding of it.
The method of synchronisation as far as i know is the line characters are bit banged into the buffer on a fifo (first in first out) basis, the response of grbl parsing and responding to a line is to send "ok"
The baud rate can be any rate - it is set in the config.ini.  My plasma table uses 9600 baud as it is the fastest speed my fibre transceivers can run at.  Using standard USB/serial comms was problematic with the plasma hf noise.  there are a couple of translators already written and available on git Grbl Panel is one such GUI which has the translator written as a subroutine within the application.  Its a shame you don't intend to write a GUI as it would be of most interest to me - ideally i would like my grbl machine to run on a GUI similar to Tormach's pathpilot.
not the play school and cartoon graphics adopted by Mach3/4.
Switching the whole machine to linux i find anally retentive.  They may be good programmers but know very little about hardware - its a constant fight.
I may not have been much help but i'm certainly interested in your project and how it progresses.

wnelis, 2020-03-02:
@MeJasonT :The remark about the serial port speed has been included in the original question.
I've looked at https://github.com/Denvi/Candle, but found only one issue there with a reference to the frequency of polling: 25 [Hz] is too high, 5 [Hz] is enough. There is no mention of problems with Grbl.

MeJasonT, 2020-03-02:
I know there is at least one polling issue with the default installation of Candle having the poll rate set to fast because i raised it.  I was also pointed towards the answer on somebody else's post

wnelis, 2020-03-10:
I've written a Python3 module which implements most of the items mentioned in my original post in a driver for Grbl v1.1. It handles the flow control, thus minimizing the change of loss of commands. The module is not complete and it has not been tested rigorously. In my test-environment it is working well, but I've not used commands like '$C' yet. See URL https://github.com/wnelis/Grbl-driver-python.