gnea/grbl #823: CoreXY Kinematics Homing Y Axis only one motor

m4rl0ne, 2020-03-03:
This is not an issue.
Can someone point me in the right direction: I have enabled CoreXY Kinematics in the latest GRBL. When I home the X Axis it works fine but when I home the Y Axis I only want the B Motor to turn, not both at the same time, otherwise it would be the normal homing routine, just with only one motor turning. I understand that this is not how CoreXY Kinematics work but I need this exact behavior for a special project. Sadly, I do not understand much about the planner or the limits.c file.
I hope you got my point and thank you very much in advance.

Pixel-Soup, 2020-03-07:
If only the b motor turns, the head will move at a 45 degree angle from the current position until the x or y limit switch gets triggered.  The head will end up at some random location.  Are you sure this is what you want?

m4rl0ne, 2020-03-07:
Yes this is exactly what I want since it's not a traditional CoreXY system I'm building but the kinematics are the same except for the one part of the homing routine