gnea/grbl #828: Dual Axis 1.1h build - Spindle enable problem

Biggs427, 2020-03-12:
I installed GRBL 1.1h with dual axis for Protoneer shield. 

The problem I now have is that the D13 pin is now for coolant but I need it for spindle enable because I don't have coolant and I control my spindle using a SuperPID.

My previous installation was using USE_SPINDLE_DIR_AS_ENABLE_PIN switch in config.h.  And it worked great. If I use this switch AND  ENABLE_DUAL_AXIS switch, will D13 pin be for spindle enable?  If not, how would I do that? 

Thanks

Biggs427, 2020-03-12:
Looking in the source code I saw that the spindle enable pin cannot be used with dual axis because Coolant Enable uses Pin D13.

I came up with the following idea:
- Make X1 and Y limit shares the same Pin (D09)
- Home each axis one after another (Z,Y then X)
- Use Pin D10 as the Spindle enable pin

To do so I came up with the following  changes in the code.  Would this make sense?  

/################  config.h Changes - Begin  ####################### 
Line 105:
// Home each axis one after another cause X1, Y shares Pin D09  AND X2,Z shares pin D12
#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)  
#define HOMING_CYCLE_2  (1<<X_AXIS) 

Line 642:
Uncomment: #define ENABLE_DUAL_AXIS
/################  config.h Changes - Begin  ####################### 

/################  cpu_map.h Changes - Begin  #######################
Line 64:
#define Y_LIMIT_BIT      1  // Pin D9 instead of D10. Shared with x1

Line 189:
#define SPINDLE_ENABLE_BIT    2  // Pin D10
/################  cpu_map.h Changes - Begin  #######################

Biggs427, 2020-03-16:
It is now working I share Limit pin X1 and Y on Pin D09 and X2 and Z on D12.

D10 is used for spindle enable .  

I had to changes config.h, cpu_map.h and grbl.h.

Here are the changes I needed:
/################  config.h Changes - Begin  ####################### 
Line 105:
// Home each axis one after another cause X1, Y shares Pin D09  AND X2,Z shares pin D12
#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)  
#define HOMING_CYCLE_2  (1<<X_AXIS) 

Line 327: (Uncomment)
#define DISABLE_LIMIT_PIN_PULL_UP  // TO have limit pins in low state by default

Line 663:  (Uncomment) 
#define DUAL_AXIS_CONFIG_PROTONEER_V3_51 
/################  config.h Changes - Begin  ####################### 

/################  cpu_map.h Changes - Begin  #######################
Line 64:
#define Y_LIMIT_BIT      1  // Pin D9 instead of D10. Shared with x1

Line 189:
#define SPINDLE_ENABLE_BIT    2  // Pin D10
/################  cpu_map.h Changes - Begin  #######################

/################  grbl.h Changes - Begin  #######################
Line 130-132:  (Comment)
  //#if defined(USE_SPINDLE_DIR_AS_ENABLE_PIN)
  //  #error "USE_SPINDLE_DIR_AS_ENABLE_PIN not supported with dual axis feature."
  //#endif
/################  grbl.h Changes - End  #######################  

It is working great and I'm now able to use my spindle with a self-squaring gantry.

/################  config.h Changes - Begin  ####################### 
Line 105:
// Home each axis one after another cause X1, Y shares Pin D09  AND X2,Z shares pin D12
#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)  
#define HOMING_CYCLE_2  (1<<X_AXIS) 

Line 327: (Uncomment)
#define DISABLE_LIMIT_PIN_PULL_UP  // TO have limit pins in low state by default

Line 361:  (Uncomment)
#define USE_SPINDLE_DIR_AS_ENABLE_PIN // Default disabled. Uncomment to enable.

Line 663:  (Uncomment) 
#define DUAL_AXIS_CONFIG_PROTONEER_V3_51 
/################  config.h Changes - Begin  ####################### 

/################  cpu_map.h Changes - Begin  #######################
Line 64:
#define Y_LIMIT_BIT      1  // Pin D9 instead of D10. Shared with x1

Line 189:
#define SPINDLE_ENABLE_BIT    2  // Pin D10
/################  cpu_map.h Changes - Begin  #######################

/################  grbl.h Changes - Begin  #######################
Line 130-132:  (Comment)
  //#if defined(USE_SPINDLE_DIR_AS_ENABLE_PIN)
  //  #error "USE_SPINDLE_DIR_AS_ENABLE_PIN not supported with dual axis feature."
  //#endif
/################  grbl.h Changes - End  #######################  

Even with my limited the code is quite simple to understand and modify to suits our needs!  

This is working for my machine but might not for another machine so use this with caution.

closed 2020-03-16 by Biggs427

LYHTSPD, 2020-03-18:
This is very similar to what I was looking to do, but I have dual axis on Y.  I think by default it says to share the dual axis limit with Z, so you could keep the Z home, and X/Y home sequence.  I was going to change the Coolant Flood (Dig pin 13) to Spindle Enable.  

I will give your changes (modified to suit my situation) a try.