gnea/grbl #832: Bug in cecksum calulation?

shaise, 2020-03-18:
eeprom.c line 133 and 144:
`checksum = (checksum << 1) || (checksum >> 7); `
Isn't it a bug? shouldn't it be logical '|' instead of boolean '||' ?
shai

langwadt, 2020-03-18:
https://github.com/gnea/grbl/issues/355

shaise, 2020-03-18:
Thanks! I had the feeling this was the reason.
I will fix it in [my port](https://github.com/shaise/grbl_cnc3040) since I'm working on a new system.

terjeio, 2020-03-18:
@shaise : you may be interested in taking a look at what [I have done](https://github.com/terjeio/grblHAL/tree/master/drivers/STM32F1xx), I have added support for quite a few additional linuxCNC G-codes such as canned cycles, the latest I have been tinkering with is G5...

shaise, 2020-03-18:
@terjeio Impressive!

shaise, 2020-03-18:
@terjeio do you have a list of all the extra features you have not present in the base 1.1f?

terjeio, 2020-03-18:
@shaise I have not made a consolidated list. There is a list of supported G-codes at the bottom of the [main page](https://github.com/terjeio/grblHAL), you need to compare that with the base. Some information can also be found in the [wiki](https://github.com/terjeio/grblHAL/wiki). There is also a table listing features present in the different ports [here](https://github.com/terjeio/grblHAL/tree/master/drivers).

One of the main differences is the architecture I have chosen, the core grbl code is made completely processor agnostic - it has to be complemented with a processor specific driver to make it work.
In addition to that I have added entry points in the HAL structure (it is function pointer based) that allows the driver layer to add (and change on the fly) functionality as it sees fit. Extra $-settings, additional reporting, additional M-codes and what not. The drivers I have made use this for stuff like SD card streaming, ethernet/Wi-Fi connectivity, extra M-codes for Trinamic support etc. All this, and more, can be done without touching the core grbl codebase.

shaise, 2020-03-18:
Thanks! That is definitely looking great.