gnea/grbl #833: grbl disconnecting/freezing at random places in a job

crazyxangel, 2020-03-20:
at first i thought it might be EMI so i made sure to get a shielded usb cable with ferrites at both ends
no dice
so i disconnected my entire machine and plugged it into my desktop pc and tried again so just UGS connected to a genuine arduino uno with nothing attached to the pins
the issue persists
so i tried CNCjs maybe UGS was acting up but CNCjs had the exact same issue
then i tried it out with UGS and an arduino due running G2core with the same type of shielded cable
and this ran just fine for multiple programs for multiple hours

zeevy, 2020-03-20:
May be you exceeded the max step rate of 30khz

crazyxangel, 2020-03-20:
how can i check if this is the case? does this happen with plenty of small movements? and how would i fix this

zeevy, 2020-03-20:
https://github.com/gnea/grbl/issues/815
Check this

crazyxangel, 2020-03-20:
that shouldnt be it then i have 40steps/mm with 5000 max

zeevy, 2020-03-20:
You can post your machine settings to get more help

crazyxangel, 2020-03-20:
$0 | 10 | Step pulse time
$1 | 25 | Step idle delay
$2 | 0 | Step pulse invert
$3 | 0 | Step direction invert
$4 | 0 | Invert step enable pin
$5 | 0 | Invert limit pins
$6 | 0 | Invert probe pin
$10 | 1 | Status report options
$11 | 0.010 | Junction deviation
$12 | 0.002 | Arc tolerance
$13 | 0 | Report in inches
$20 | 0 | Soft limits enable
$21 | 0 | Hard limits enable
$22 | 0 | Homing cycle enable
$23 | 0 | Homing direction invert
$24 | 25.000 | Homing locate feed rate
$25 | 500.000 | Homing search seek rate
$26 | 250 | Homing switch debounce delay
$27 | 1.000 | Homing switch pull-off distance
$30 | 1000 | Maximum spindle speed
$31 | 0 | Minimum spindle speed
$32 | 0 | Laser-mode enable
$100 | 40.000 | X-axis travel resolution
$101 | 40.000 | Y-axis travel resolution
$102 | 800.000 | Z-axis travel resolution
$110 | 5000.000 | X-axis maximum rate
$111 | 5000.000 | Y-axis maximum rate
$112 | 500.000 | Z-axis maximum rate
$120 | 1000.000 | X-axis acceleration
$121 | 1000.000 | Y-axis acceleration
$122 | 1000.000 | Z-axis acceleration
$130 | 200.000 | X-axis maximum travel
$131 | 200.000 | Y-axis maximum travel
$132 | 200.000 | Z-axis maximum travel

zeevy, 2020-03-20:
I think it will excede 30k when both x and y are movef with G0 command (in such case your mm/min becomes around 7k

crazyxangel, 2020-03-20:
hmm i'll just try running it at 2500 then see if it holds up like that i'll let you know in a couple hours

crazyxangel, 2020-03-20:
on ugs it still froze (but the entire ugs program froze so still waiting to see if cncjs works well)

crazyxangel, 2020-03-20:
CNCjs has been running for an hour now without issue probably too early to say it is all fixed but its working way better than before thanks for the advice bout the max mm/min @zeevy  will keep testing CNCjs trough the night/tomorrow and close if all goes well

abdset, 2020-03-21:
This issue have nothing to do with UGS or GRBL but hardware side issue:

- Use shielded wires for your motors and connect the shield only to arduino GND (Motor side should not be connected to anything).
- Same for limit switches if you are using them.
- Arduino motors drivers must have a heatsink, without that they get too hot and cut themself from power (Safety feature).
- Do not connect power supply GND with arduino GND since some cheap power supplies causes issues in this case.
- If the power supply and the arduino are in the same metal coated box, put a metal plate in between and connect that to arduino GND (Basicly the arduino should be separated from power supply or spindle power cable when they are in same box).
- Do not "ziptype" arduino wires with any kind of power wires unless they are cable shielded inside the box and that shield should be connected to arduino GND.
- I recommend using DRV8825 motor drivers instead of A4988.
- Power the motors with at lease 4 Amps.
- Always use a Fan to cool down the motors drivers (They get too hot even with heatsink installed).
- Poor USB cable quality can also cause this issue (It happend to somebody before) don'nt trust your cable, get new one.
- Check the arduino usb connector if nothing is broken, i'm talking about the square piece of plastic that holds the 4 usb contacts in the center of the connector (This happend to me, i had to ressolder a new usb plug).

crazyxangel, 2020-03-21:
I'm literally running the arduino without anything connected to it like I stated in the issue itself I am also using a shielded USB cable as also stated in the issue so I highly doubt my motor drivers psu's nor my cables have anything to do with it seeing as they are disconnected laying in the garage while it is just the naked arduino connected to my computer that is displaying this issue

abdset, 2020-03-21:
"connected to my computer"
Are you using raspery PI or a PC ?

crazyxangel, 2020-03-21:
the issue persisted on both raspberry pi and my desktop pc

abdset, 2020-03-22:
The issue persist, then its usb cable or ...........
Are you using a clone version of arduino ?
Many people report bugs on a cloned arduino.
You said the error shows up even when using only arduino with the PC when nothing else is connected, then the only possible causes i see is the usb cable or the arduino itself.
I also have seen some people reflash the arduino bootloader to fix connexion issues, you may try that.

abdset, 2020-03-24:
Check this:
https://github.com/gnea/grbl/wiki/Known-Issues