gnea/grbl #834: Coolant Enable Quickly Toggled on Power-up

BusbyGT, 2020-03-21:
I just completed a controller build using the latest committed version of GRBL with the dual axis homing enabled (Protoneer CNC v3.51 option) and am having an issue with the coolant output.

When I power up, the coolant output is quickly toggled on and off. It also toggles on and off when I open UGS but I am assuming that may be due to UGS issuing a reset. This causes my IOT relay to quickly cycle, at least once, and turn on my shopvac.

The output works normally using the M8 and M9 command just fine, it's just a strange boot-up issue.

I am assuming this is just a start-up state for D13 being one thing until the dual axis config takes over the output pin but have yet to confirm that.

I'm trying to find a config.h setting that would correct this but haven't been successful yet.

Also, on a related note, the commit log on GitHub has the step/dir ports reversed from how the code has it.

terjeio, 2020-03-21:
Typically pins are configured as high-Z inputs on startup, at least for the processors I use. The relay driver electronics should be designed to handle that.

BusbyGT, 2020-03-21:
I should also add that this relay has worked perfectly fine on GRBL v1.1 on the regular coolant output before I upgraded my controller. I'm also using the same relay for my spindle/router and am not having the same issue with that output, it's only the coolant enable output. I will try to reverse the polarity of the relay inputs to see if that changes anything.

BusbyGT, 2020-03-21:
Changed polarity and got no relay activation at all which I figured would happen.

It is cycled three times when powered on.

I looked at a lot of config.h options and I didn’t see anything that would fix the cycling. I was planning to dive more into the power on sequence to see if I can narrow it down.

abdset, 2020-03-22:
Try to add electrolyte capacitor to that pin to prevent it from enabling it at the start up.
The value depends on how long that pin stay enabled at startup.
Begin with 10 V / 100 uF, if not working use 10 V / 470 uF.
You may put a diode then the capacitor to prevent current going back into arduino when capacitor is discharging after pin voltage drops to 0.
[image]

BusbyGT, 2020-03-22:
i understand you could wire in a capacitor but there’s other drawbacks to doing that and that’s not the point here. There’s something in the code that is causing this to cycle when enabling the dual axis homing feature. The spindle PWM output is the same style output and it doesn’t have this problem. I have the same style relay connected to both and actually tried both relays on each output and only the coolant enable output had this problem.

BusbyGT, 2020-03-23:
I've found the issue and it's the Arduino Uno bootloader.

It's setup to toggle Pin 13, normally an LED output, on boot-up and when the Uno is programmed.

It appears the issue is resolvable by reprogramming the bootloader to disable the LED flashing, which won't toggle D13.

Doesn't seem like D13 is a good choice for any outputs due to this.

Now I get to rethink my approach to this and how to fix it.

BusbyGT, 2020-04-15:
I ended up flashing a new boot loader that disables the led flashing on boot and it solved the problem. I'll close this issue and hope that maybe having it out there will help somebody else.

closed 2020-04-15 by BusbyGT