gnea/grbl #837: Jog cancellation (0x85) sometimes not working

shaise, 2020-03-22:
Hi,

I saw the Issue #95 thread and I know it was closed, but this issue is still a problem (happens also in the Mega version) 
I think I have new insights, though. And it happens with a single jog cancel, no further jog commands sent.
Here it what happens: when the jog cancel is sent, the cnc slows to a stop and then starts again going much farther then then a a single jog segment sent before. 
The jog is done by continuously sending a 0.25mm motion, each time 'ok' is sent back. all messages are ceased after the cancel command is sent. 
I have investigated it quite a bit, and here is what I found out: if I send the cancel before the last jog command "ok" received, it triggers the issue. If I wait for the ok and only then send the cancel, everything works great. 
It looks like that if the cancel is done before the ok received, indeed the machine stops and all active buffers are purged. However, there is still one pending planner command in the belly (perhaps waiting for a vacant buffer) that executes and causes this issue.
To recreate the issue, send many jog commands of say .25mm, until ok is not returns (buffer full), then immediately send the cancel. 
I'm still digging into Grbl code, hopefully I find where the problem is.

Many thanks for this great project!
shai.

AlexHolden, 2020-03-22:
This has been discussed a few times and the old discussions are hard to follow, but it sounds like you figured out the answer for yourself. After you have started sending a jog command, you need to finish sending it and wait for the OK response to come back before you can safely send the jog cancel realtime command. Otherwise a race condition can occur where the cancel command is processed the instant it is received, then shortly afterwards grbl finishes processing the jog command and starts jogging again.

AlexHolden, 2020-03-22:
I just remembered there is another jogging quirk to be aware of. Immediately after sending a realtime jog cancel command, send "G4P0" and wait for it to be OKed before you send anything else. I forget the exact reason why, but Bad Things can happen if you don't wait for the cancel operation to complete before sending new movement commands.

shaise, 2020-03-22:
OK, I definitely found the bug in the code on motion_control.c, mc_line function:
There is a busy wait function when the buffer is full:
do {  
    protocol_execute_realtime(); // Check for any run-time commands  
    if (sys.abort) { return; } // Bail, if system abort.  
    if ( plan_check_full_buffer() ) { protocol_auto_cycle_start(); } // Auto-cycle start when buffer is full.  
    else { break; }  
} while (1);  
As soon as the cancel command is sent, the buffers are free and mc_line quits this busy-wait and immediately issues a new line...
I will try to fix it, but its best be done by the owners.

terjeio, 2020-03-22:
@shaise : Both the serial input and block input buffers needs to be cleared on jog cancel. My attempt to fix it for the Mega version can be found [here](https://github.com/terjeio/grbl-Mega).

shaise, 2020-03-22:
@terjeio ,

I have fixed the problem. Clearing serial input and block buffers is not enough. The code is simply waiting for all buffers to clear, and immediately issues the pending command - this command is in the code and not in any buffer. This is cross thread synchronization issue.

abdset, 2020-03-22:
I figured out how to jog properly without using 0x85 code:

- Lets say you want jogging feedrate to be  1000 mm/min, you need to cut that to 100 milliseconds sections.
- 1000 per minute is 1000/60 = 16.666 per second
- 100 ms is 1sec/10
- 16.666 / 10 = 1.666 per 100 ms (this is the value of the jogstep, it moves 1.666 mm each 100 ms).
- Now you send those steps each 100 ms when jog button still holded down.
- When jog button is released the move will stop immediately cz jogstep(1.666) and jog time (100ms) are synchronized on feedrate value.
- The 100 ms timing is very good, i dont recommend to think changing it but you can play with the feedrate value and apply the formula to it.

**The final formula in code is:**
      feedRate = 1000;
      jogStep = feedRate / 600; 
      Serial.println("$J = G91 G21 X" + jogStep + " F" + feedRate);
      delay (100);

This worked like a charme in my machine.

abdset, 2020-03-22:
I forgot to ask:
How do you guys read "ok" status from grbl? Any code example ?

shaise, 2020-03-22:
for anyone who is interested, [this is my fix](https://github.com/shaise/grbl_cnc3040/commit/b220e351408db5ea9767512366fc4751d60ae814)
@abdset this was my original solution, and indeed it is good enough. But I did want to have an option to cancel any motion.

sadezohdi, 2020-06-01:
> for anyone who is interested, [this is my fix](https://github.com/shaise/grbl_cnc3040/commit/b220e351408db5ea9767512366fc4751d60ae814)
> @abdset this was my original solution, and indeed it is good enough. But I did want to have an option to cancel any motion.

Thanks for your idea.
The machine's performance has improved. But the problem is not completely solved.

TrevorTighe, 2021-03-29:
I can confirm that the Cancel file transfer is being sent before the OK is received on my version of UGS.
Universal Gcode Platform 20201202
On a windows 10, 64 bit machine
Here is two examples of the jog commands being sent

This one caused the jog to continue jogging after stopping when I release the jog button
ok
>>> $J=G21G91Y-8F4000
ok
>>> $J=G21G91Y-8F4000
**** Canceling file transfer. ****
ok 

Where as this one stopped correctly

ok
>>> $J=G21G91Y-8F4000
ok

**** Canceling file transfer. ****

So it seams that GRBL has the problem where a cancel can cause an unpredictable state.
Also maybe UGS should wait longer for the OK before sending the cancel.

The over run usually only happens if the jog button is held down for a long distance and at high feed rate.
But some times it can happen on a short jog.

On long jogs, the issue happens about one in 5 times.

Is there a new version of GRBL or UGS that fixes this issue?
It is very disconcerting when the router stops but then continues on after I release the jog button.

abdset, 2021-04-05:
I added a 6 milliseconds delay right before sending 0x85 command and it did fix the problem for me, never happend again,
you have to put the delay after the joystick has been released so it does not loop it.

_if joy not centered
send jog command
else
send a delay
send 0x85_

(I use an ESP32 to drive the joystick whitch sends commands to arduino uno via serial connectors of the boards, this allows me to use a joystick and send the job via bluetooth, grbl 1.1h is on the arduino and custom code for joystick and data bridging PC/Arduino is on the ESP32).
Machine <--> arduino uno <--> ESP32 (Joy & BT/serial bridge) <--> PC

I attach my esp32 code if someone is interrested (The IDE is PlatformIO installed on Visual studio code)
[main.txt](https://github.com/gnea/grbl/files/6256610/main.txt)
PS: Use notepad++ to view it and select C on language tab if you don't have any IDE installed yet.

Usefull link for PlatformIO installation: https://www.youtube.com/watch?v=JmvMvIphMnY&ab_channel=DroneBotWorkshop

Hope this helps.