gnea/grbl #838: "HEIGHTMAP", Z-axis moves into wrong direction

bud86, 2020-03-23:
Hi,

I´ve set up GRBL to move the z-axis up for the home position. If I use the z-probe command, the z-axis moves downwards until a closed circuit is connected (crocodile clips connected to "A0" on my woodpecker board).
But if I want to create a heightmap the z-axis moves upwards until hitting the limit switch.

I tried to research which command I have to change, with no success.
Appreciate any advice!

bud

MeJasonT, 2020-03-23:
Are the homing directions correct
if not you will need to change settings
$3 to get machine moving in correct direction, its effectively a binary flag ie 111 +1+2+4 representing X,Y,Z in most cases the default build of grble works for most people 000 - so if you need to change a direction say Y axis would be 010 (decimal 2)
The same for the homing setting $23
if they are correct then
Did you home your machine first before attempting to probe
if your machine thinks z is somewhere other than the home position it will do weird things.
let us know how you get on.
stay safe and well dont get the lurgy

MeJasonT, 2020-03-23:
I assume you dont have the door alarm set in config and no door switch fitted.  It can be a royal pain in the bum as it retracts the z and if not switch is fitted can be spuradic.  the default grbl build config is disabled/not active.  false triggering could also be an issue, If so have you put a capacitor across the probe input pin and ground.

bud86, 2020-03-23:
Wow, I`m amazed and ashamed at the same time, how simple the solution is. Asking if I homed my machine before probing brought me to it.
Indeed I homed my machine but changed the position afterwards before starting the heightmapping. Setting XY and Z not to 0 was my failure!
Thank you so much for that bump in the right direction 👍 👍 👍 

One more thing I`d like to ask. Is it possible to change the z-feedrate and how far it moves down?
Maybe $24 is also used for the feedrate during heightmapping?

MeJasonT, 2020-03-23:
A typical Z probing operation would be
G21 ------------------  Set to metric, (or Imperial if you are that way inclined)
this is good practice you have no idea what state the last NC/Job file had left your machine in
$H  ------------------  Home the machine - no need to program move if youer machine is already homed

G92 Z0  --------------  Set z to 0
G01 Z-20 F100  ----- Feed down to -20 mm 100mm per min
F50 -----------------  Slow things down to 50mm per min.
G38.2 Z-60 -----------  Send the spindle down max travel eg. -60mm to zero tool tip.
G92 Z0 ---------------  Set switch point to 0

Zero of Z is now set

G01 Z5 ---------------  Move spindle up 5mm to release switch (

F50 ------------------  Slow things way down to make an accurate zero point.
G38.2 Z-7 ------------  Send the spindle down to -7mm
G92 Z0 ---------------  Set switch point to 0
Or to set a work height
G92 Z??? -------------  Offset the switch point Z to the actual height of my table Z.

G01 Z30 F500 ------ Raise the bit to 30mm above my surface.

Try the commands with your hand on the reset switch just in case, when you see what they are doing and the behaviour is as expected try the next line.  It will give you confidence when you see it work both in yourself and the machine. - trust me i did it this way.