gnea/grbl #839: $0=?  tb6560 and nema23

hitchhiker85, 2020-03-24:
hi! i play arround with feedrates and microsteppings...

1/8 microsteps (1600) and a feedrate of 270 is the highest i can get on my leadscrews (m6 with 1mm pitch) befor the steppers starts to make bullshit :)

for me i dont understand $0!

the tb6560 has a minimun clock pulse width of 30 microsteppings in my setup.

is this the value for $0=? or i am wrong?

thanks!

Norwegian-Gadgetman, 2020-03-24:
Yes, $0 is the parameter that should be set to 30, or more...

Okay...

1600steps/rotation(also steps/mm) and a speed of 270mm/minute, I get... 4.5mm/second, = 7200microsteps/seconds.
That number is way below the 30000 that GRBL can handle, so that's not an issue.
I think that converts to about 138uSeconds per pulse, so 30uS shouldn't be too long for it, either.

Now, what does it do that's 'bullshit'?

And can we get some of the other settings?

Have you tried increasing the current?
Also, leadscrews aren't high speed systems. you may be bumping up against a limit there?

Trygve

hitchhiker85, 2020-03-24:
current is set to 2.5A and the nema23 is a 2.6A stepper

the accleration is set to 30

the steppers stalls.

i think i must let the feedrate at 250 270mm/min and live with it!

the problem is the leadscrew, its what you said!

but what happend if i set $0=30 and what is when i set it to $0=10?
with a feedrate of 250mm/min and microstepping 1/8 = 1600?

nothing will change for me? i dont understand!

thanks for your reply and help!

Norwegian-Gadgetman, 2020-03-24:
I'm assuming that $100 - $102 is set with 1600, and that $110 - $112 has a max speed set that is higher than 270?

($100/$110 = X-axis, $101/$111 = Y-axis, $102/$112 = Z-axis)

Which voltage do you run it at?
12V is probably not going to be useful. 24V is better, but if you can get 36 or even 48V that would help a lot.
(12V rating is kind of a silly thing, particularly if it runs at higher than a few steps per second. It needs either time or a bit of power to produce the torque)

hitchhiker85, 2020-03-24:
i forget to say:

i use a 32v with 5a  supply 

decay setting at 50%

hitchhiker85, 2020-03-28:
me again!

with feedrate 250mm/min all ok
above 270mm/min steppers stalls and make high freq. noise.

what can i do?