gnea/grbl #841: this is hard to explain laser not functioning grbl 1.1f or 1.1h only grbl.9

quotelawrence, 2020-03-25:
the issue happened a while ago I made the mistake of updating grbl 1.1f, it had a prompt saying this could damage grbl board but I thought why would they have a function that would update yet damage so I hit the update but than the laser module stopped working, X & Y still work the Z still functions but when I plug in the grbl 1.1f & up boards (i have a couple) there is no operation to the laser, the light on the board lights up like a signal is being given but the laser does nothing, I then plugged in the grbl .9 board and played and rather than using a 255 maximum or 1000 maximum I entered 700 laser strength it started working but when I go to the grbl .9 parameters there is no $30= $31= $32= it stops at $28= does anyone know how to get the boards to once again function properly

JBSchueler, 2020-03-27:
Dit you check the pinning file?
The wiki can tell you all about it : [Connecting Grbl](https://github.com/grbl/grbl/wiki/Connecting-Grbl)

2 pins are swap since grbl1.0.
> **For Grbl v0.9 with variable spindle PWM ENABLED:** (NOTE: The Z-limit and the spindle enable pin are swapped, because we had to access the hardware PWM on D11 for variable spindle PWM output to work.) We are still updating this pin configuration at the moment by weighing future options. We'd like to only change the pins once. Stay tuned!

quotelawrence, 2020-03-27:
thank you for your response I am not sure about pinning though this morning i noticed on a different engraver cnc i own the wires running from the board to the module maybe running in a different pattern red black blue from board and into module black red blue, i am starting to think that chinese wire harness are subject to error this pattern works magnificently on one machine though I have to modify harness on the other ,

abdset, 2020-03-28:
You may take a look here:
https://github.com/craftweeks/grbl-1.1f.customized-for-laser/blob/master/grbl/cpu_map.h

quotelawrence, 2020-03-28:
Ha ha ha I do not do programming

abdset, 2020-03-28:
its not realy programming, there is some setting inside the file you need to change values, you dont need to changes any instruction inside, just the values.
Think about your grbl settings $1, $2 and so on.
Its the same here but look more complicated just because they have names instead of $1, $2.

quotelawrence, 2020-03-28:
however I did find the wiring was incorrect and I cut and spliced and re-hooked it is operating though there is another issue as the laser is activated it starts at a lower heat and after a while start to generate and focus to a higher heat, I have adjusted the laser to the best point I can though once running the point seems to open up and the heat is lost in the grbl setting can any of the $= be changed to increase heat or to cause higher frequency ?

JBSchueler, 2020-03-29:
The change the power of the laser it has to be connected to a PWM output.
That was the reason why the pinning was changed since FW1.0.
If the laser pin is not connected to a PWM outpout, it is just on or off.

quotelawrence, 2020-08-17:
thank you I noticed the wiring was incorrect and changed that to match my other laser wire color code, it has been running though the new issue is (it appears as if the heat varies) the start burn is hot, but then it seems to not stay hot, so the laser is on yet I will get a cut, it is soon not burning and i get dots rather than constant lines, and laser heat