gnea/grbl #843: Resume after Pause behaves irrational

deHarro, 2020-03-29:
Hi!
I use OpenCNCPilot GUI (OCP) together with GRBL V1.1f.

OCP has an option to pause the streaming of a gcode file.
When streaming gcode is paused, one can issue jog commands.
After that one can resume the streaming job. 
The Gcode in the file has absolute coordinates.

What happened:
I paused the gcode job, the GRBL buffer is depleted, the milling bit sits 8 mm deep in the workpiece.
I jog to Z somewhere above the workpiece to have a look on what the bit has done so far to the workpiece.
After that I issue "resume".
The bit is lowered to its previous height and the job went on.

Next time, same procedure, "pause", issue jog commands for Z but then jog to X somewhere to positive X values, so I can even better see what has been done so far.

Now the problem:
After issuing "resume" the bit is lowered to its former Z position (into the workpiece at the current, displaced, X position) and then it moved to the coords which are given be the next gcode command... ruining the workpiece.

My expectation was, that first the X and Y coords would be restored, after that Z should be restored.

Is there a means of parametrizing that?
Or is it mandatory to not jog around during a running gcode streaming job?

Thanks!

Harald

abdset, 2020-05-01:
- I have tryed the same and i was surprised to see that  OpenCNCPilot let you jog when a job is in progress (even paused), this is not safe.
- In other softwares (like UGS platform sender) when you pause a job, jog commands are locked.
- You may pause the job when the spindle is up away from workpiece so you dont need to move it manualy.
**Last word:** jogging during the job progress is not safe until something is added to grbl to improuve it so it does allow jogging and go back at the position before the pause when resume is used and for now this is not done yet and grbl get confused when you do that.

deHarro, 2020-05-02:
Are there chances that this "something" will be implemented in the future?

Harald

hairu, 2020-05-02:
From your description, Grbl is doing exactly what OpenCNCPilot told it to do.  The "pause in streaming" you described was simply OpenCNCPilot stopped submitting commands to Grbl.  When you resumed, OpenCNCPilot started sending commands again.  Unfortunately, you had changed the state by jogging the machine to a new location, but Grbl has no idea that you want to go back to the original location/state because Grbl was not aware of the "pause", only OpenCNCPilot was.  If there is state you would like to be restored after the resume, it would need to be OpenCNCPilot that does it.

deHarro, 2020-05-02:
Ok, I seemingly misunderstood the principle of the two distinct queues (streaming and jogging, respectively) GRBL maintains.

I try to get the save/restore managed by OpenCNCPilot, it has a macro feature, it should be feasible.

Thanks for the clarification!

Harald

closed 2020-05-02 by deHarro