gnea/grbl #845: Experimenting tool changer with code functions

MeJasonT, 2020-04-01:
Im wanting to have a digital pin wait for an input (pulseIn) say HIGH and once the signal changes back low I want to change the state of the receiving pin to an output and then transmit a pulse.

The Idea is to use 1 digital pin like a handshake between an Arduino running GRBL and a Tool changer
the idea is to have GRBL transmit a pulse of a fixed duration say 2 seconds (not important just long enough for the tool changer to recognise  it).

The receiving Arduino (Pin) then changes from an input to an output and then transmits a continues pulse until a boolean state is changed in code which switches the output off again.

The GRBL Arduino would be coded to transmit only the 2 second pulse then switch to listen.

This way if a tool change is required Grbl transmits a signal then reverts to listening to the pin as long as the signal is High for example then it knows the tool changer is busy.  Once the tool changer has completed its task it drives the signal low then becomes an Input pin again.  On seeing this pin change GRBL continues with its milling operation.

The UNO version of GRBL only has 1 spare pin which happens to be A4.

Overall Plan.

What the overall project might look like is that the serial commands sent to grbl will be bridged to a second usb using a virtual serial port application.  The tool changer arduino will listen for tool change instructions in transmitted gcode and parse out the Tool commands M6 and T* (* = the tool number)
This cant instigate a tool change as the values read into the GRBL buffer means that it is not finished all of its operations just because it has changed a tool change instruction and would need to instigate the operation when it was ready - The tool changer needs to tell GRBL its finished.
As the code advances the M6 T* gcode will tell the tool changer arduino what tool its expected to load so when it receives the go/ active pulse it will use that information to offer the correct tool.
The rest will be sensors for extend and retract of the changer motors for extend and retract, sensors to see tool clear and tool loaded, limit switches and a tool location ID sensor (check tool is aligned with spindle and not half cocked. Possibly an encoder to offer belt and braces for the tool number/location, Once homed the stepper should be able to know where it is but if it lost steps may not offer the correct tool and carnage will happen.
The reset pin will be daisy chained with the grbl arduino so any alarm will kill the operation on both Arduinos.
@chamnit  I would be most interested to have your input, in particular the stepper motor code.  My plan is to try using accelstepper as i dont know how to strip and use the highly efficent and fast GRBL stepper code/routines.  The ARM build has still not appeared to become a huge DIY project with build your own hardware, so im sticking with the Atmega 328/Uno version for now.  Every time im searching for a new ARM version i come across those either working on code for a newer more different processor and not managing to run with one until all users have exhausted it like in the case of the UNO.  Or it means buying a development board for a new processor and still not having any hardware guidance.  The good news is Linux programmers are still anal and think the parallel port still has a future, Think i upset them asking for a USB version. lol.

uChip, 2020-04-01:
Sorry for jumping in here unasked.

What you describe is certainly doable.  A similar mechanism is used on the ATmega in the I2C interface and other places.  It works by turning the output into an "open-collector" mode.

To avoid problems with outputs fighting, the mechanism is usually set up as an active low signal as follows (on both controllers).

	set the data output register to "0"
	turn on the pin's internal pull up resistor (or use an external one)
	send the desired signal to the DDR control for the pin
		a "1" on DDR turns the output off - pull up will pull output high
		a "0" on DDR turns the output on - data output register "0" drives output low

Then the protocol is:		
1. The GRBL controller pulls the output low (set DDR to "0") for X amount of time.
2. The tool change controller must respond to the low signal by also pulling its output low (set DDR to "0").  The tool change controller must respond within X time.
3. After time X the GRBL controller releases the signal (set DDR to "1") and reads the data (on the pin, not the data out register, nothing special here, that's how the ATmega works) and waits as long as signal is low.  (there needs to be a timeout here that aborts the job if the tool change controller does not come back soon enough)
4. When the tool change completes the tool change controller releases the output (set DDR to "1") to be pulled up high again.
5. The GRBL controller continues operations when it sees the data pin go high.

Note the controllers work in an inherently noisy electrical environment.  The internal pull ups may not be sufficient to maintain a high level in all that noise.  A stronger external pull up (and/or capacitive filter) may be required.

This scheme depends on no other part of the GRBL code touching the DDR register for that pin.  If it is "unused" then nothing should, but I am not familiar enough with the code to say for sure.

Good luck with your project.

Fading back to lurker mode...

MeJasonT, 2020-04-01:
Ok, intense,   
You are not jumping in my friend, you are most welcome.
Im off reading up on the subject, thanks for the info.

terjeio, 2020-04-01:
My 32-bit port has support for an optional tool table and for hooking ATC code into the grbl core:

https://github.com/terjeio/grblHAL/blob/cdab733cae58b598ecc849a01008a5157ad93bcd/grbl/hal.h#L173-L174

The core handles parsing of T and M6/M61 commands and dispatches calls to the entry points as needed - no "man in the middle" required. I have been tinkering with an ATC for a while, using a separate controller for handling the mechanics and I2C for issuing commands/reading back statuses.

uChip, 2020-04-01:
That may be a better solution.  A brief search in the code shows Arduino A4 to be COOLANT_MIST or COOLANT_FLOOD depending on whether dual axis mode is enabled.  Either way, A4 seems to be in use (cpu_map.h for Protoneer CNC shield).  Alternatively, to use A4 for ATC these other functions would need to be disabled so that they don't interfere.  Another alternative would be to disable I2C and use one of those pins for the one-wire handshake protocol.

uChip, 2020-04-01:
Whups.  Strike that last alternative.  I forgot A4 and A5 are already shared with I2C so that's literally the same pins either way.  And A5 is used for a Probe input so it's not available either.  So the bottom line is that all the pins are already allocated to something.  You will need to make a trade off of what to give up to get ATC functionality.

MARIOBASZ, 2020-04-18:
Also on the way to my atc. I lack knowledge to fully understand what they write.
Using the pause pin or the probe pin I think that is what I will use to adjust movements and communication between grbl and atc