gnea/grbl #847: How do I install GRBL updates?

jlo726, 2020-04-05:
New to this world of grbl I need help to do grbl updates. I just purchased a "Universal Engraver" 15000mW Blue. I was told it's a Eleks maker laser machine. I cannot get the laser to turn on and off. It just on continuously. I have the update: grbl_v1.1h.20190825 (1).hex, however I have no idea how to put it in the controller. I have XLoader and I have Arduino. Now how do I "install" and then use LightBurn as my software. LaserWeb came with the engraver but seems a lot more complicated.

jlo726, 2020-04-07:
I have everything ready to go in Arduino, so when I connect to laser I assume I need to be online. Do I need to be online when ever I'm making something???

abdset, 2020-04-07:
Some gcode senders needs to be online to work, some others dont.
Working offline software for example: UGS Platform, but you will need to install java first to make it work.
Download UGS Platform here:
http://winder.github.io/ugs_website/download/
I'm using nightly build version and it seems to be quiet stable to me.

jlo726, 2020-04-07:
Got this error code. Arduino: 1.8.12 (Mac OS X), Board: "Arduino Uno"

In file included from /Users/JMBP/Documents/Arduino/libraries/grbl/grbl.h:43:0,
                 from /Users/JMBP/Documents/Arduino/libraries/grbl/config.h:30,
                 from /Users/JMBP/Downloads/grbl-1.1h.20190825/grbl/examples/grblUpload/grblUpload.ino:3:
/Users/JMBP/Documents/Arduino/libraries/grbl/nuts_bolts.h:51:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 
In file included from sketch/grblUpload.ino.cpp:1:0:
/private/var/folders/_y/6wfs25696hq2hpwxbm54qg000000gn/T/AppTranslocation/EA015413-EFFE-4C24-884A-EC8CB8F4AE8B/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 
In file included from /Users/JMBP/Documents/Arduino/libraries/grbl/grbl.h:43:0,
                 from /Users/JMBP/Documents/Arduino/libraries/grbl/config.h:30,
                 from /Users/JMBP/Downloads/grbl-1.1h.20190825/grbl/examples/grblUpload/grblUpload.ino:3:
/Users/JMBP/Documents/Arduino/libraries/grbl/nuts_bolts.h:52:0: warning: "min" redefined
 #define min(a,b) (((a) < (b)) ? (a) : (b))
 
In file included from sketch/grblUpload.ino.cpp:1:0:
/private/var/folders/_y/6wfs25696hq2hpwxbm54qg000000gn/T/AppTranslocation/EA015413-EFFE-4C24-884A-EC8CB8F4AE8B/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:92:0: note: this is the location of the previous definition
 #define min(a,b) ((a)<(b)?(a):(b))
 
In file included from /Users/JMBP/Documents/Arduino/libraries/grbl/grbl.h:43:0,
                 from /Users/JMBP/Documents/Arduino/libraries/grbl/config.h:30,
                 from /Users/JMBP/Downloads/grbl-1.1h.20190825/grbl/examples/grblUpload/grblUpload.ino:3:
/Users/JMBP/Documents/Arduino/libraries/grbl/nuts_bolts.h:55:0: warning: "bit" redefined
 #define bit(n) (1 << n)
 
In file included from sketch/grblUpload.ino.cpp:1:0:
/private/var/folders/_y/6wfs25696hq2hpwxbm54qg000000gn/T/AppTranslocation/EA015413-EFFE-4C24-884A-EC8CB8F4AE8B/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:124:0: note: this is the location of the previous definition
 #define bit(b) (1UL << (b))
 
Sketch uses 29684 bytes (92%) of program storage space. Maximum is 32256 bytes.
Global variables use 1484 bytes (72%) of dynamic memory, leaving 564 bytes for local variables. Maximum is 2048 bytes.
In file included from /Users/JMBP/Documents/Arduino/libraries/grbl/grbl.h:43:0,
                 from /Users/JMBP/Documents/Arduino/libraries/grbl/config.h:30,
                 from /Users/JMBP/Downloads/grbl-1.1h.20190825/grbl/examples/grblUpload/grblUpload.ino:3:
/Users/JMBP/Documents/Arduino/libraries/grbl/nuts_bolts.h:51:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 
In file included from sketch/grblUpload.ino.cpp:1:0:
/private/var/folders/_y/6wfs25696hq2hpwxbm54qg000000gn/T/AppTranslocation/EA015413-EFFE-4C24-884A-EC8CB8F4AE8B/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 
In file included from /Users/JMBP/Documents/Arduino/libraries/grbl/grbl.h:43:0,
                 from /Users/JMBP/Documents/Arduino/libraries/grbl/config.h:30,
                 from /Users/JMBP/Downloads/grbl-1.1h.20190825/grbl/examples/grblUpload/grblUpload.ino:3:
/Users/JMBP/Documents/Arduino/libraries/grbl/nuts_bolts.h:52:0: warning: "min" redefined
 #define min(a,b) (((a) < (b)) ? (a) : (b))
 
In file included from sketch/grblUpload.ino.cpp:1:0:
/private/var/folders/_y/6wfs25696hq2hpwxbm54qg000000gn/T/AppTranslocation/EA015413-EFFE-4C24-884A-EC8CB8F4AE8B/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:92:0: note: this is the location of the previous definition
 #define min(a,b) ((a)<(b)?(a):(b))
 
In file included from /Users/JMBP/Documents/Arduino/libraries/grbl/grbl.h:43:0,
                 from /Users/JMBP/Documents/Arduino/libraries/grbl/config.h:30,
                 from /Users/JMBP/Downloads/grbl-1.1h.20190825/grbl/examples/grblUpload/grblUpload.ino:3:
/Users/JMBP/Documents/Arduino/libraries/grbl/nuts_bolts.h:55:0: warning: "bit" redefined
 #define bit(n) (1 << n)
 
In file included from sketch/grblUpload.ino.cpp:1:0:
/private/var/folders/_y/6wfs25696hq2hpwxbm54qg000000gn/T/AppTranslocation/EA015413-EFFE-4C24-884A-EC8CB8F4AE8B/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:124:0: note: this is the location of the previous definition
 #define bit(b) (1UL << (b))
 
Sketch uses 29684 bytes (92%) of program storage space. Maximum is 32256 bytes.
Global variables use 1484 bytes (72%) of dynamic memory, leaving 564 bytes for local variables. Maximum is 2048 bytes.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
[... cut ...]
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

abdset, 2020-04-07:
I dont have mac but i was checking your error and i recommand you to delete grbl folder from Documents/Arduino/libraries/ since its already in examples folder.
Seems like declarations are being read twice from different files, you may clean up all files that concerns grbl but keep the one in examples folder.
Did you put grbl folder into more than one location ?

In the other hand, i noticed that the following line in config.h is not commented by default.
#define DISABLE_LASER_DURING_HOLD
So i think if you use the HEX file with XLoader may work just fine.

Hope this will help you fix it.

jlo726, 2020-05-19:
Are there any videos that could walk me through all of this. It’s so confusing to me.

abdset, 2020-05-19:
This link shows how to do it using Arduino IDE (NOT the Xloader program):
https://www.youtube.com/watch?v=m998bYioHqs
In the video he is skiping the part where you should connect your arduino to usb and do some settings before upload the grbl.
In the following link, you will skip to 3:13 time and see how to connect arduino to its IDE and do the settings for ardduino uno and usb port:
https://www.youtube.com/watch?v=zlRCzGwHft0
In the second video he is not showing true files and his arduino is not connected but the explaination should be enough for your needs.

If you want to flash with Xloader using HEX file then begin the second video from the start.

Hope this helps.
Good luck.

jlo726, 2020-05-20:
I think you were right there are too many copies of things in the console. How do I clear it out and start over? Also, do I need to buy the Arduino board? Don’t I just use the USB to plug into my board? I appreciate your help.
Josie

jlo726, 2020-05-20:
Does this make sense to you and can you tell me why I’m having so many problems? 
Josie

Waiting for connection...
Waiting for connection...
Grbl 1.0b ['$' for help]
Your Grbl may not support Variable Power mode (M4)
If your laser does not turn off between cuts,
switch to the GRBL-M3 device
Homing
error: Setting disabled
ok
ok
ok
ok
ok
<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000,SC:0>
ok
Starting stream
[Pgm End]
Stream completed in 0:00
Waiting for connection...
Waiting for connection...
Port opened, waiting for response.
Grbl 1.0b ['$' for help]
Your Grbl may not support Variable Power mode (M4)
If your laser does not turn off between cuts,
switch to the GRBL-M3 device
<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000,SC:0>
ok
Starting stream
error: Undefined feed rate
On or near line 4:
[Pgm End]
On or near line 6:
Stream completed in 0:00
ok
Waiting for connection...
Waiting for connection...
Grbl 1.0b ['$' for help]
Your Grbl may not support Variable Power mode (M4)
If your laser does not turn off between cuts,
switch to the GRBL-M3 device
Homing
error: Setting disabled
Waiting for connection...
Waiting for connection...
Grbl 1.0b ['$' for help]
Your Grbl may not support Variable Power mode (M4)
If your laser does not turn off between cuts,
switch to the GRBL-M3 device
Homing
error: Setting disabled
Waiting for connection...
Waiting for connection...
Grbl 1.0b ['$' for help]
Your Grbl may not support Variable Power mode (M4)
If your laser does not turn off between cuts,
switch to the GRBL-M3 device
Homing
error: Setting disabled
$$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$30=1000. (rpm max)
$31=0. (rpm min)
$32=1 (motor lock bool)
$33=7 (motor mode mask:00000111)
$100=80.000 (x, step/mm)
$101=80.000 (y, step/mm)
$102=80.000 (z, step/mm)
$110=10000.000 (x max rate, mm/min)
$111=10000.000 (y max rate, mm/min)
$112=10000.000 (z max rate, mm/min)
$120=250.000 (x accel, mm/sec^2)
$121=250.000 (y accel, mm/sec^2)
$122=250.000 (z accel, mm/sec^2)
$130=500.000 (x max travel, mm)
$131=500.000 (y max travel, mm)
$132=500.000 (z max travel, mm)
ok
Waiting for connection...
Waiting for connection...
Port opened, waiting for response.
Grbl 1.0b ['$' for help]
Your Grbl may not support Variable Power mode (M4)
If your laser does not turn off between cuts,
switch to the GRBL-M3 device

abdset, 2020-05-20:
- Speaking of flashing grbl into the board, you are supose to have an arduino boad already, but still i don't know the model of your engraver to tell whitch board it has.
- Speaking of homing error, if your machine has limit switchs, you may set $21 and $22 to 1, to do so, type in console:
$21=1
$22=1
Those two lines will activate homing using limit switchs to tell the machine where the limits are, if you are not sure about limit switchs, it is then safe to keep them disabled with 0 value (at least for now).
- Please provide us with the brand and the model number of your engraver so we can check on the net its specifications and provide you better help, or post a picture of it where we can see brand and model number (On some stick or so).

abdset, 2020-05-21:
Was checking on the net, and found some usefull links for you to try out:
https://forum.lightburnsoftware.com/t/laser-does-not-turn-off-between-cuts/1046
https://forum.lightburnsoftware.com/t/turning-laser-off-between-cuts/2306