gnea/grbl #848: Help on customizing the firmware

gisep, 2020-04-06:
Hello guys,
more than one time i tought to experiment some customization on the firmware, but i sincerely don't know where to start. I can write some basics in arduino, i'm not an expert, but it's ok with taking my time and studying the necessary to do something.

1) How can i understand the structure of the grbl firmare in order to know how to modify it?

I can imagine that the grbl code (and others too) is done by assembling "blocks", so you can have a hierarchical and synthetic seen to get oriented and operate for sectors in specific subfiles.
It would be GREAT to have some guide/tutorial on how to modify the firmware, by doing it in blocks (as plugins). This would improve a lot the potential contribute of all the community for the customization and the growth of the firmware, since that every request can be developed by anyone or any team!

I know that the first limit is the used ROM, but this could be overpassed by customizing the features depending on the board: Arduino UNO, Arduino MEGA, WeMos D1 and all the others. The second limit is the minimum number of input-output ports needed to run the firmware, and here comes the second question:

2) What is the actual number of analogic and digital ports used by grbl? what are them?

Then, here comes my third and direct request that could be a practical example, probably easy enough to start with:

**How can I change the two "feed hold" and "resume" buttons to one unique push button?**
I've got an external 2€ thermostat called W1209 (reading a probe in the laser case) with a switch that automatically goes on when the temperature limit is overpassed, and returns off when it returns cooler. Attaching it to the Arduino, it will do an automation of the pause-resume cycle based on temperature control. This change would also free one port (the resume one) that could be used, for example, to control a light source or any other thing.

What do you think?
I will be glad for any answer.

Giuseppe

zeevy, 2020-04-06:
https://awesome.tech/grbl-demystified/
This might be helpful for you

gisep, 2020-04-06:
Thank you!
I will put an eye in the next days

gisep, 2020-04-07:
Hello,
I have seen that post, that explicitates the structure of grlb functionment. But it doesn't talk about the programmation of the firmware, that is what I need to.

Has someone got some skills to show how to navigate in the firmware to find the commands assigned to a port? (for example how to find the "feed hold" section and change the switch to a "push" button?

abdset, 2020-04-12:
I'm not a good programmer but as far as i know:

- GRBL firmware is made in few parts in source code, each of them has specific job to do, they are files.
- Usualy the file name explains what that part is doing.
- Each file has its own variables whitch are named the way they explain what they are about.
- You will see include fonction on the top of each file meaning the named file (included one) properties can be used by current file (its a connetion between files or parts if you wish) so you can call a procedure or a fonction of the included file and even external variables values.

Meaning:
- You need a programmer skills to understand C or C++ instructions.
- You need an IDE that can open all those files at once, those files together are called "a project", the IDE will check for errors, build the project (HEX file in this case).

Quick way for newbie:
Use notepad++ to open all the files at once and find what you are looking for, THIS IS NOT THE CORRECT WAY TO DO IT since you have nothing to check your code like C or C++ IDE and may lead to crashs.

I have been working on C++ and i still noob at it, i'm just trying to give you the very basics things.