gnea/grbl #849: Please make a close loop cnc grbl for precision

3636yatin147, 2020-04-09:
Changing in grbl library for add a feed back(close loop stepper motor)  to measure a rotation of stepper motor for precision using a encoder. Please any one can make this change it is very important.

cri-s, 2020-04-09:
It don't make any sense doing it on the grbl controller.
I use 8pin uc that receive step dir, encoder input and outputs step/dir to
the driver.

Norwegian-Gadgetman, 2020-04-09:
If it's so important, and anyone can make the change, why don't you do it yourself?
And while you're at it, feel free to rewrite it for a microcontrolelrboard with more memory also, because the Arduino Uno is practically full. There's not enough storage to fit anything like that.
GRBL as it is has already broken everyone's ideas of what can be crammed into such a small controller.

biasedlogic, 2020-04-10:
> Changing in grbl library for add a feed back(close loop stepper motor) to measure a rotation of stepper motor for precision using a encoder. Please any one can make this change it is very important.

No, it isn't important, and it's the wrong way to do it. Grbl is motion controller, not servo amplifier. The function of following position is one to implement in servo amp and there are such ones out there that do exactly that.
What *could* be asked for, is for grbl to send absolute positions rather than incremental signals to servo amplifiers over some bus, like rs485, but that may be hard to implement (although I don't say it's not possible, at the cost of temporal resolution)

ASHIT-AXAR, 2020-04-10:
there is nothing to do with GRBL.
this ability should be available in your stepper/servo driver.

[image]

there is a ALARM signal in the driver you can route back to arduino. (either to PAUSE or RESET button)

3636yatin147, 2020-08-30:
Thank you sir, it is very helpful for me ! but can you show me the circuit
about (there is a ALARM signal in the driver you can route back to arduino.
(either to PAUSE or RESET button))