gnea/grbl #852: limit switch trigger report

jon-daemon, 2020-04-15:
I use grbl 1.1f. I have read everything about limit switches and decided to build a pcb with optocouplers. I have connected them as normally closed in serial. I use a DC to DC isolation converter on board to isolate the 5V rail on the limit switches but the ground is common with the arduino. I still get random triggers from the limit switches. I use shielded wire but I haven't connected the shield to the ground. Should I connect the shield to the ground on both the limit switches and steppers? (the cables are next to each other) Could a 100nF capacitor in paraller with the limit switch input help?
Here is my board. The transistors are there to invert the optocoupler output and light an led when the switch is pressed.
[image]

I notice that grbl doesn't report which limit switch is triggered. I just got an alert. Is there any option to enable more detailed reporting when I get an alert and debug more easily my problem?

biletnikov, 2020-04-15:
It sounds like you have issue with noise protection in your board. Do you have a schema of it ?

abdset, 2020-04-15:
- Speaking of grounding wires, yes you can connect grounding shield to arduino (Controller side) but do not connect it to anything in the other side (Motors or limit switchs side).
- Speaking of capacitors, yes you need to add capacitors in parallel of limit switchs, better be inside the controller box and add 4.7 Kohms resistors connected one side to + limit switch wire and the other side to +5V of arduino, here is a picture how to do so but dont follow resistors colors in the picture cz they are wrong (The picture is for normal open switchs configuration):
[image]

- Finaly, you need to check the voltages of the optocouplers when the limit switchs are not triggering to be sure they are not reaching the limit to light up the diodes inside. You can make a test by pushing a limit switch with your finger and see if they are triggering only when they sould so. Having a look at the optocouplers datasheet will help you to know at what voltage they will trigger.

jon-daemon, 2020-04-15:
I followed the exact same schematic [here](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches). I just didn't use two separate power supplies. I use the same 5V rail from the Arduino with an isolation module(B0505S). My mistake is that I use the same ground from the Arduino to the output of the isolation module. I will try to cut the traces on my board and separate the ground for the limit switches side.

The optocouplers work fine when I press the limit switches. Where should I add the capacitors and the resistors? The schematic above doesn't use optocouplers. Should they be on the Arduino side or the limit switches side of the optocouplers?

I use NC connection and I might have a loose connection which triggers the switch. I only had a few random triggerings of the limit switches but I don't know which switch. My main question is how should I know which switch is triggered. Is there any option in config.h to enable more detailed report in alarms?

biletnikov, 2020-04-15:
The current limit resistor + Optocoupler (input side) which you have is a good noise protection schema. However, switch have bouncing issue, which can be fixed with RC. In your case, I would suggest to add 0.1 uF capacitor in parallel to the contacts of switch.

And, check carefully your B0505S module, which may cause the problems.

jon-daemon, 2020-04-17:
I have 3 transistors, one for each led to check the operation of the switches before starting any job on the CNC. They are on the output side of the optocoupler.

I checked the B0505S with an oscilloscope and it seems fine. I also checked the input of the optocoupler. There is bouncing from the buttons but the main problem appeared when I started moving the steppers. There is so much noise which eventually could trigger the optocoupler. I added one 100nF cap in parallel with the switch input on my board and the noise almost disappeared. Here are some screenshots from the oscilloscope.

machine idle
[image]

moving one axis slowly
without capacitor on switch input
[image]

with capacitor
[image]

I will fix the boards (add capacitor and separate ground) and I will order a new batch. I could sell them as bare boards for anyone interested to check it out instead of ordering a whole batch. Of course they are open source and I will publish the schematic and gerber files on my website if you want to order it yourself. I will inform you when they will be ready.

biletnikov, 2020-04-17:
Ok, it would be good if you share the schematics.

Thanks.

jon-daemon, 2020-08-15:
I have built a new board for the limit switches.
Schematics and gerber files are available [here](https://hackermagnet.com/limit-switch-board-for-cnc-machines/).
You can also order a bare pcb, source and and solder the components yourself based on the BOM on my website. (Sorry for the high shipping cost. There is no cheaper way to send it from Greece)