gnea/grbl #853: GRBL as base for industrial servo controller

jomiq, 2020-04-16:
I'm working on a two-axis stage for a small high-speed plotter using integrated BLDC servo motors like these: [GYEMS RMD-L-40](http://www.gyems.cn/794904.html).
The servomotors are going to be controlled using a position/speed protocol over CAN bus. Each servo is running a PI control loop and can report cumulative position of the 14-bit encoder, as well as motor current, speed, and temperature. Closed loop speed and torque control modes are also possible. There are current, voltage and temperature protections built into the drivers. In essence, it is an industrial servo interface in a tiny package. (I believe the intended main application is stuff like camera gimbals.)

I'd like to adopt GRBL to generate real-time command sequences for this system. At the moment I have not decided what controller to use. Ideally I'd like to target something more powerful than the atm328/arduino, but performance-wise it should not be a problem with a low-end microcontroller as long as it can attend to the bus interface in a timely fashion. 

I believe a generalized g-code interface to integrated servos should be of great benefit to the community as a whole. I'm therefore asking if there is any work being done in this direction, and also for collaborators and/or suggestions for how to proceed. 

Cheers

terjeio, 2020-04-16:
I have made a HALified port where a quite a bit more data than step/direction is passed to the hardware driver. I am currently using this to implement spindle synced motion, but perhaps it could be used (as a starting point?) for servos as well?

doppelhub, 2020-05-12:
Grbl could certainly be a starting point, but what your suggesting would require a pretty substantial rewrite to planner.c.  Also, you'd certainly need a more powerful microcontroller (ideally that had a CAN bus); the 328p is running full tilt.  My Grbl variant uses every single flash byte and every single 328p subsystem... I should have migrated to a more powerful system ages ago, but with over 10,000 units deployed that's a non-starter at this point.