gnea/grbl #854: ENABLE DUAL AXIS

JohnLagou, 2020-04-16:
hello 

ver 1.1h

// Select the one axis to mirror another motor. Only X and Y axis is supported at this time.

#define DUAL_AXIS_SELECT  X_AXIS  // Must be either X_AXIS or Y_AXIS

NO ERROR ON UPLOAD
-----------------------------------------------------------------------
if select 

 Y_AXIS

#define DUAL_AXIS_SELECT  Y_AXIS  // Must be either X_AXIS or Y_AXIS

ERROR ON UPLOAD

Arduino: 1.8.12 (Windows 10), Board: "Arduino Uno"

C:\Users\...\libraries\grbl\report.c: In function 'report_realtime_status':

C:\Users\....\Arduino\libraries\grbl\report.c:654:1: error: expected declaration or statement at end of input

 }

 ^

exit status 1
Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

revyradness, 2020-04-26:
The latest release had a typo that has since been fixed [here],(https://github.com/gnea/grbl/commit/bdc2da9b5972c5972cdd4a4f045a575f09ff74cc) that was causing your problem.

Download [this](https://github.com/gnea/grbl/archive/eefe2bb95bb7b21ec2bb87e6ab6e20747e1626c4.zip) which includes the commit that fixed it. I've tested it and it works great.

JohnLagou, 2020-04-27:
Thanks a lot yes it works