gnea/grbl #857: Servo not working on CNC Shield V4

EranIDar, 2020-04-25:
Hi !

I'm creating a CNC Plotter with 2 Bipolar Steppers for X & Y and a servo as a Z axis for the pen.

I bought a CNC Shield V4 ([link](https://fr.banggood.com/Geekcreit-CNC-Shield-V4-With-Nano-3_0-A4988-Stepper-Motor-Driver-Board-For-Arduino-products-that-work-with-official-Arduino-boards-p-1250215.html?rmmds=myorder&cur_warehouse=CN)) on banggood. **Disclaimer : don't buy this board,  it's full of bugs.**

I installed grbl on the Arduino Nano and did the following things : 
- In cpu_map.h : Invert the pinout between X_STEP_BIT, Y_STEP_BIT Z_STEP_BIT and X_DIRECTION_BIT, Y_DIRECTION_BIT, Z_DIRECTION_BIT  (as I said this board is garbage 😂, they even failed for that)
- In config.h : comment VARIABLE_SPINDLE to disable it

So I have connected the servo on Z+ which is connected to pin 11 of the Nano. And in Universal Gcode Sender, I enter the commands "M3" or  "M5" but the servo is not moving at all. I tried with grbl 1.1f and 1.1h

I've already tested my servo on the pin 11 of a spare Arduino Nano, it works fine.  

Do you have any ideas or something I have missed...

Thank you,

EranIDar, 2020-04-25:
Here are my parameters : 

$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)

ardmed, 2020-04-27:
hi guys me to have the same problem , first time working well
but now nor working 
grbl-servo-master
cnc shield v3
2 motors onw servo SG09

EranIDar, 2020-05-08:
I managed to make it work !!!

Because I tried several repo (grbl, grbl-servo, Grbl Pen Servo) and I added this libraries into Arduino there was a mix of files (really weird). So I removed all the other grbl libraries except Grbl Pen servo and it worked like a charm !

closed 2020-05-08 by EranIDar