gnea/grbl #858: Limits issue

ameisso, 2020-04-26:
Hello everyone 

I try to add limit switch to my plotter. 

I use a [laserAxe](https://www.google.com/imgres?imgurl=https%3A%2F%2Fimages-na.ssl-images-amazon.com%2Fimages%2FI%2F61p1rHjt5EL._AC_SX522_.jpg&imgrefurl=https%3A%2F%2Fwww.amazon.com%2FEngraving-Cutting-Machine-Engraver-Controller%2Fdp%2FB07QL7C7F8&tbnid=YaGvkBPPZUO5vM&vet=12ahUKEwjQsqOSuobpAhVPgRoKHQi-AUYQMygKegUIARCEAg..i&docid=Y4PHE8NxPYdbZM&w=522&h=327&q=laseraxe&client=firefox-b-d&ved=2ahUKEwjQsqOSuobpAhVPgRoKHQi-AUYQMygKegUIARCEAg) controller with only two axis

On this board, limits are wired to pin 9, 10 and 12 
so I changed changed cpu_map.h like this : 
```
#define LIMIT_DDR DDRB
#define LIMIT_PIN PINB
#define LIMIT_PORT PORTB
#define X_LIMIT_BIT 9   // Uno Digital Pin 9
#define Y_LIMIT_BIT 10   // Uno Digital Pin 10
#ifdef VARIABLE_SPINDLE // Z Limit pin and spindle enabled swapped to access hardware PWM on Pin 11.
#define Z_LIMIT_BIT 12   // Uno Digital Pin 12
#else
#define Z_LIMIT_BIT 12 // Uno Digital Pin 11
#endif
#define LIMIT_MASK ((0 << X_LIMIT_BIT) | (0 << Y_LIMIT_BIT) | (0 << Z_LIMIT_BIT)) // All limit bits
#define LIMIT_INT PCIE0                                                           // Pin change interrupt enable pin
#define LIMIT_INT_vect PCINT0_vect
#define LIMIT_PCMSK PCMSK0 // Pin change interrupt register
```

I've also changed config.h so it dont try to move Z : 
`#define HOMING_CYCLE_0 ((1 << Y_AXIS) | (1 << X_AXIS)) `

when I boot up the card with cncjs, I got an alarm. So I hit $H and the axes moves in the right direction. 
When I close both limit switch, nothing happens, the axis continue to move and never stop. 

Note that for now, my switchs are 10cm wires that I shortcut with GND to close them. So I don't think that could be noise related

Here is my $$ output : 

```
> Grbl 1.1h ['$' for help]
> client> $$
> [MSG:'$H'|'$X' to unlock]
> $0=10 (Step pulse time, microseconds)
> $1=25 (Step idle delay, milliseconds)
> $2=0 (Step pulse invert, mask)
> $3=3 (Step direction invert, mask)
> $4=0 (Invert step enable pin, boolean)
> $5=0 (Invert limit pins, boolean)
> $6=0 (Invert probe pin, boolean)
> $10=16 (Status report options, mask)
> $11=0.010 (Junction deviation, millimeters)
> $12=0.002 (Arc tolerance, millimeters)
> $13=0 (Report in inches, boolean)
> $20=0 (Soft limits enable, boolean)
> $21=1 (Hard limits enable, boolean)
> $22=1 (Homing cycle enable, boolean)
> $23=3 (Homing direction invert, mask)
> $24=10.000 (Homing locate feed rate, mm/min)
> $25=500.000 (Homing search seek rate, mm/min)
> $26=5 (Homing switch debounce delay, milliseconds)
> $27=1.000 (Homing switch pull-off distance, millimeters)
> $30=1000 (Maximum spindle speed, RPM)
> $31=0 (Minimum spindle speed, RPM)
> $32=1 (Laser-mode enable, boolean)
> $100=100.000 (X-axis travel resolution, step/mm)
> $101=100.000 (Y-axis travel resolution, step/mm)
> $102=100.000 (Z-axis travel resolution, step/mm)
> $110=10000.000 (X-axis maximum rate, mm/min)
> $111=10000.000 (Y-axis maximum rate, mm/min)
> $112=10000.000 (Z-axis maximum rate, mm/min)
> $120=1000.000 (X-axis acceleration, mm/sec^2)
> $121=1000.000 (Y-axis acceleration, mm/sec^2)
> $122=1000.000 (Z-axis acceleration, mm/sec^2)
> $130=1000.000 (X-axis maximum travel, millimeters)
> $131=1000.000 (Y-axis maximum travel, millimeters)
> $132=200.000 (Z-axis maximum travel, millimeters)
```

I'll be really grateful for any insight ! 

Best

doppelhub, 2020-05-12:
Sounds like something isn't hooked up correctly.  Grbl uses interrupts to determine if a limit switch has tripped.  With everything connected, connect a voltage meter (DMM) up to the limit switch pins... if the voltage is ~5 volts in one limit switch position and ~0 volts in the other, then Grbl will see that state change.  If you don't see that voltage swing when tripping the limit switches, then something isn't hooked up correctly.

botio, 2020-11-15:
```
#define LIMIT_DDR DDRB
#define LIMIT_PIN PINB
#define LIMIT_PORT PORTB
#define X_LIMIT_BIT 1   // Uno Digital Pin 9
#define Y_LIMIT_BIT 2   // Uno Digital Pin 10
#ifdef VARIABLE_SPINDLE // Z Limit pin and spindle enabled swapped to access hardware PWM on Pin 11.
#define Z_LIMIT_BIT 3   // Uno Digital Pin 12
#else
#define Z_LIMIT_BIT 3 // Uno Digital Pin 11
#endif
#define LIMIT_MASK ((0 << X_LIMIT_BIT) | (0 << Y_LIMIT_BIT) | (0 << Z_LIMIT_BIT)) // All limit bits
#define LIMIT_INT PCIE0                                                           // Pin change interrupt enable pin
#define LIMIT_INT_vect PCINT0_vect
#define LIMIT_PCMSK PCMSK0 // Pin change interrupt register
```

Maybe try this? since u use "PINB" the number u used can't be 9,10.....

ameisso, 2020-11-17:
Hello, 

Thanks for the feedback. 
On the board I had offered connectors on 9,10 for limit switch 
And I read [here](https://www.arduino.cc/en/Reference/PortManipulation) that these where on portB ? Did I miss something. 

BTW, i switched control board for a more convenient one, so everything is ok for me now. 

Thanks

closed 2020-11-17 by ameisso