gnea/grbl #859: Missing wiki page

arkypita, 2020-04-26:
Look like Laser mode page is disappared

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

arkypita, 2020-04-26:
Url bring me to "create new page". 

[image]

pastprimitive, 2020-05-05:
Same issue, the laser wiki page is missing.

johndudleyhayes, 2020-05-08:
The configuration page also seems to be missing.

[image]

neilferreri, 2020-05-08:
I was able to get a cached version of the config page, but I have no idea how to restore from HTML.
[grblConfig.zip](https://github.com/gnea/grbl/files/4602018/grblConfig.zip)

arkypita, 2020-05-09:
I think I was able to restore them from a previous version.
I don't know why they was deleted.

[image]

[image]