gnea/grbl #860: Config of Step pin only for dual Y-Axis motor

gxdeange, 2020-04-27:
Hi Team,

I have a custom board build in mind and only need the configuration of the additional motor STEP pin for the second Y-Axis motor.
I believe that A4 is not necessarily used and was curious to know how that would be configured in cpu_map file. All other pin assignments can stay the same as there is a second limit switch on the Y-Axis that is looped in with the first Y-Axis limit switch.
Also out of curiosity, if the homing sequence is configured to only home one axis at a time will this home Y motor 1 and then Y motor 2?
I apologise if this does not make sense as I am not a coder and I suspect I will mess it up quite badly if I touch it.

Cheers
G

doppelhub, 2020-05-12:
An easier way to do this is to connect an unused pin (e.g. A4) to the "sleep" pin on your stepper controller.  I'm suggesting this because the Step pins need to stay in sync with each other, which is a more difficult task.  It sounds like you're trying to perform an auto-level feature, in which case you can just put the stepper driver to sleep.  If this will work for you, then that function is one line and basically sets/clears the "sleep" pin.

If you want to make sure the pin is enabled during homing, just add a line to enable the stepper to the homing routine (mc_homing_cycle).

I have this implementation in my grbl variant, and use it to turn one stepper off during an auto-level routine that uses two limit switches.  It works great.

gxdeange, 2020-05-12:
Hi doppelhub,

Thanks for your response. I have a cnc machine that has dual motors for the Y-Axis which requires both motors to be in sync and running at the same time, step and direction.
I am not sure if the ganged motor feature will work in this scenario as both Y motors would need to home until one switch is triggered and then each one would have to home independently and trigger their own switch.
This is the machine I have that I use for testing if that would provide more context into seeing if independent homing would work.
https://www.makerstore.com.au/product/osm-wb50x75/

Cheers
G

neilferreri, 2020-05-12:
@gxdeange I might be missing something, but why not just share the Y step signal with both Y motors/drivers?

gxdeange, 2020-05-12:
Hi Neilferreri,

Currently all my designs have the option for a Clone driver to be either X, Y or Z. I suspect that my understanding of what the ganged motor option is used for might be incorrect.

gxdeange, 2020-05-12:
Thanks doppelhub and neilferreri for your comments.

closed 2020-05-12 by gxdeange