gnea/grbl #861: Limit Switches triggering Alarm at startup

zachrockafellow, 2020-04-27:
Hi, 
I'm having some issues wiring my limit switches up with GRBL

I've got a 3018 CNC Router setup with GRBL 1.1f (using a Woodpecker v3.4 control board), and wired/programmed to use limit switches ($21=1) When $21=1, I can't use the machine because the alarm goes off every time I start it up - even when no limit switches are plugged in. I expected to plug in my switches, activate the setting in grbl, and I'd be able to move my axes around using the controller, as I've done with other identical boards.

Right now, with the value of $21 set to 1, my offline controller won't move any axes at all.

I tried deactivating the pull-up resistor by editing the config.h file, but the alarm still goes off every time, until I deactivate the switches ($21=0)?

What steps should I take to diagnose the exact issue, and does anybody have an idea as to how I might be able to fix this? I've got another woodpecker 3.4 board that has none of these issues

abdset, 2020-05-01:
- What are you using to move the machine when its offline ?
- You may check $5 value according to your switchs wiring (Normal OFF or normal ON), if they are normal off keep $5 to 0 , if they are normal ON set the value to 1.
- Also you may upgrade to grbl 1.1h, it handles offline jogging better.
- You say the alarm is OFF when you start it, i dont understand the meaning of this because with grbl when you start the machine and $21 is 1 the alarm should be ON meaning the machine should be homed (safety feature) you can kill the alarm by sending soft-reset and $X then you are supose to home the machine.
- Keep in mind in arduino based controllers, Z limit and spindle pins have been inverted since grbl 1.1, to put them back like they were before you need to edit config.h and set variable spindle to OFF (just comment the line to disable it).
Please put a feedback if you manage to fix it, it can help others.
Wish you best lock.

doppelhub, 2020-05-12:
Sounds like a noise issue.  You need to add a simple RC lowpass filter to the limit switches.

otoyasan, 2020-06-03:
3018pro使ってます。grbl 1.1fです。
オフラインコントローラーのSDカードに$Hと書いたファイルを置いて、初めにそれを実行してから使用してます。

JackyLau, 2020-10-23:
Same issue here

Kamoba, 2021-07-18:
like said @abdset make sure you match your $5 setting to your switch wiring.
$5=1 for NC switch
$5=0 for NO switch

For better reliability  check out the noise filtering wiring.
[https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches)