gnea/grbl #862: Using decimal number for $100, $101 and $102 - odd Wpos as a result

humneraudio, 2020-04-28:
Whenever I enter a decimal point number as one of the step/mm values, GRBL seems to calculate the movement weirdly. I don't have a better way to describe it. I'll try and explain.

Lets say I set X axis steps/mm as follows
$100=400.5
I then zero out X axis
I will then send it to 
X1
It will report Wpos x as 1.001
X2
it will report Wpos x as 2.0
X1.3
it will report Wpox x as 1.301
X7
it will report Wpos x as 6.999

After making these movements, I then zero out the X axis and it will zero it out to 0.001

If i reset GRBL and then zero out X it will zero out to 0.0

For many of the movements it can't seem to resolve to a whole number. Which I thought was odd as I'm telling GRBL to go to these whole numbers.

I came across this as my x and y axis is ~0.5mm off over a 400mm distance (I'm using cheap lead screws) and I wanted to finely adjust the steps/mm to compensate and after I dialed in my values(which are decimal numbers) I noticed the Wpos being odd. 

It occurs in bCNC, Candle and grblControl

It also happens in 0.9grbl and 1.1grbl

PicEngraver, 2020-04-29:
Grbl uses integer math routines, not floating point.  It is likely that is why you are seeing the very small discrepancies, which are smaller than a hobby level cnc machine can adjust to, anyway.

terjeio, 2020-04-30:
Stepper motors cannot move fractions of a [(micro)step](https://hackaday.com/2016/08/29/how-accurate-is-microstepping-really/), eg. with your setting 401 step pulses is sent to the motor for 1mm movement. 1.0 / 400.1 * 401 = 1.001 (when rounded to 3 significant digits). Assuming metric this is a 1 micrometer error, likely less than the [accuracy](https://tech.thk.com/en/products/pdf/en_a15_011.pdf) of the mechanical parts.