gnea/grbl #863: Loss of position on resume after Safety Door feedhold

neilferreri, 2020-05-08:
I'm trying to track down an issue I'm having with feedhold parking / resume.
Basically, on resuming after the parking is completed, I'm losing about one mm (Z ends up too high).  I am retracing 5mm at a feed of 100mm/min and then parking at Machine Z-5mm at 500mm/min.
My Z is a ballscrew that has no issues with backlash or lost steps, and those feedrates are well below my max.  This is a very repeatable issue.
If I run the same motion through a macro or similar, I do not see the same loss of Z position.  I can run 
```
G91
G1Z5F100
G90
G53Z-5F500
```
and back to Z0 all day without any loss of Z position.
I seem to have narrowed it down to the RESUME where the position is lost, not on the actual parking.
Any thoughts?

doppelhub, 2020-05-12:
To make sure I understand your issue:
A: While you're running the four lines above, you're sending ! (feed hold), followed by ~ (cycle start).
B: When you perform 'A', the machine loses 1 mm of position?
C: If you repeat this four times, are you now off 4 mm?
D: How are you measuring how much travel you're losing?  Are you measuring against a fixed point?

neilferreri, 2020-05-12:
The four lines I ran were to simulate the parking behavior, to test if this is a mechanical issue. I was able to run that macro and back to Z0 repeatedly with no loss of position.
The issue occurs when I use the safety door switch to issue a feedhold and park and then resume. To be honest, I didn't measure, but I was losing about 1mm per park/resume cycle.  I have parking set to the grbl defaults which will park the Z 5mm below the limit switch. After running a few cycles I was contacting the switch.
Again, this loss of position only happens on the park/resume cycle and specifically is happening on the resume.

neilferreri, 2020-05-13:
I'm open to any suggestions.
Tests I could run?

neilferreri, 2020-06-05:
Issue was related to the way my controller handles the pin I was using for a resume.  The pin was originally assigned to A0, and there's a hardware logic ic built-in to disable the steppers when grounded.  Back to `~` for me on that machine.

closed 2020-06-05 by neilferreri