gnea/grbl #866: soft limits reversed behavior

MoustafaElkady, 2020-05-16:
Hello,
Thanks for a great GRBL its awesome.
I'm using 1.0.9
these are my $$

`$0=5 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=20.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=2000.000 (x max rate, mm/min)
$111=2000.000 (y max rate, mm/min)
$112=2000.000 (z max rate, mm/min)
$120=500.000 (x accel, mm/sec^2)
$121=500.000 (y accel, mm/sec^2)
$122=500.000 (z accel, mm/sec^2)
$130=50.000 (x max travel, mm)
$131=50.000 (y max travel, mm)
$132=50.000 (z max travel, mm)`

first problem
when i click on + or - to move X or Y and the unit is mm they move 5mm not one

second problem
l set soft limit as the above, when i trying to increase X or Y to test soft limits, it "ALARM: Soft limit" when it reach 0, that's mean its not moving at all because 0 is the start point

and when i'm trying to move it in reversed way -Y or -X it movese to 50 and 50 correctly but in the second direction.

i need to set limits in a positive X and X

FrankBolleri, 2024-03-03:
Hello all, I have documented a difference between two popular Amazon modules based both on GRBL that seems to me connected with the "second problem" of El-kady:

[image]
[image]
[image]

I can note that 4 Axis have a more recent version of 1.1f (20230316).
I also note that the 3 axis, that is more old (20170801), have a clear indication "CUSTOM".

As stated in the last screenhoot... **which is the correct behaviour?**

And a final question: as far a someone here know, is possible to re-flash the firmware of these Amazon's boards, maybe using Arduino IDE?

Thanks a lot in advance
Francesco

breiler, 2024-03-03:
@FrankBolleri the difference between these two controllers is that the "4 axis with double Y" firmware was compiled with the option `HOMING_FORCE_SET_ORIGIN` (https://github.com/gnea/grbl/blob/master/grbl/config.h#L129). It is by default disabled.

This can be seen in the output from the $I command where it is reporting the OPT flag Z. (https://github.com/gnea/grbl/blob/master/grbl/report.c#L370)

FrankBolleri, 2024-03-05:
> @FrankBolleri the difference between these two controllers is that the "4 axis with double Y" firmware was compiled with the option `HOMING_FORCE_SET_ORIGIN` [...]

Thanks a lot for the answer.