gnea/grbl #867: just got latest version 1.1h, but my arduino say 0.9j !!

OZ2CPU-thomas, 2020-05-17:
so it is not the lates ??

neilferreri, 2020-05-17:
You have to flash the new firmware to the Arduino. How did you do that?

OZ2CPU-thomas, 2020-05-17:
downloaded the new zip, went into its folder for programming, its was an ino file, so it opened my arduino, then then I hit program, it compile, ok, uploads, and I go to serial monitor,
and it is version 0.9j i got..

neilferreri, 2020-05-17:
You need to replace the grbl folder in your libraries folder with the new one that you extract from the zip.

OZ2CPU-thomas, 2020-05-17:
oh, i did not do that.. bummer..

neilferreri, 2020-05-17:
Don't forget, your $$ settings will be overwritten.  Make a backup.

OZ2CPU-thomas, 2020-05-17:
i did make a back up, of my $$ setting, 
in fact I took a brand new arduino, just to be sure
so I can run them in parallel. thanks mate, lets call this one closed, i was a but jerk :-)

closed 2020-05-17 by OZ2CPU-thomas