gnea/grbl #868: Simulator available for `grbl`?

rirze, 2020-05-20:
Hi,

I'm currently using a fork of this project for one of my machines. I would like to add onto it using a unit test based workflow.

I was very interested when I found [grbl-sim](https://github.com/grbl/grbl-sim) but I found that it was [about to be depreciated](https://github.com/grbl/grbl-sim/issues/13#issuecomment-296378563) back in 2017. 

Is this still true? Is there possibly another method to simulating grbl?

I would appreciate any kind of response or update. Thank you again.

neilferreri, 2020-05-20:
I always have an extra Arduino laying around.  You can load grbl and use the bare Arduino as the simulator.

rirze, 2020-05-20:
I see that, but I'm specifically trying to create a unit test framework that be run on CI/CD workflows. I can't expect a Cloud service to have an Arduino, for example.