gnea/grbl #869: Homing in X direction doesn't work

cong-99, 2020-05-21:
grbl version: 1.1h on Woodpecker 3.2 controller. 

$20=0
$21=0 
$22=1

Odd problem - I recently put in a set of limit switches, set up the homing dir mask, etc.  Sending a $H command starts the whole process correctly, with the Z axis movement.  As soon as that completes, the machine stops and seemingly hangs (does not respond to any further command inputs via serial, does not produce any messages on the serial terminal). 

I modified config.h to enable separate X, Y and Z homing and discovered that Z and Y homing work correctly, while X homing always produces a "stuck" machine (steppers powered, controller unresponsive to any commands). The only way out of this seems to be a hard-reset. 

X movement using something like G1 X25 F100 works properly. 

I have tried:

- swapping A4988 driver modules (just for luck) - no effect
- testing with hard limits ($21=1) - this works as expected, i.e. 
G1 X 50 F250   gets the X axis moving, and the moment I manually trigger the X limit switch, the system stops and an alarm message is produced. 
- doing a $HX, and triggering the X limit switch by hand to simulate that the head unit "reached" the limit switch, etc - but the controller never comes out of its seemingly locked state (no "ok" message). 
- checking if it's false-triggering from the limit switches (they are configured as NO - I added 2.2K pull-up resistors to ensure there's little chance of noise ingress; there's already an RC network on each of the pins) - seems clean.  
 
In short - it's only the X-homing that doesn't work, and anything that "calls" the routine results in a controller hang. 

I will be happy to try out any mods (software or hardware) to solve this problem, i.e. could this be a solder bridge issue, causing an interrupt that is only allowed through during X-homing?  I might be able to find this if I knew which interrupts / pins to look at...

PicEngraver, 2020-05-21:
I do not know about the Woodpecker 3.2 board, but I have a 3.4 board, and the X and Z switch inputs are marked in reverse.  IE: the X is marked Z, and Z marked X.  Just a thought.  

A picture of your board would help to see if the 3.2 board is marked the same.

cong-99, 2020-05-21:
Thanks for your comment/input!  I believe the switches are wired correctly, as evidenced by hard limit operation  (tested in all 3 axes), i.e. 

G1 X50 F250

Then as the head moves, I press the X-limit switch: does stop.  

I had a quick look using a DMM to verify that the XEND pin is connected to pin 13 of the ATMEGA-328P (PB1, aka D9 of an Arduino).. yup, it is....

PicEngraver, 2020-05-21:
Maybe you have already seen this.  But do hope it helps.
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

To check your switches, short one of them and start up PicSender.
After PicSender finishes sratup, click the '$X' button to clear the alarm.
Then click the '?' button.  You should see a message like this: **<Idle|MPos:0.0000,0.0000,0.0000|FS:0.0,0|Pn:X>**  with the **Pn:X** indicating the 'X' switch is made.  The 'X' will change to 'Y' or 'Z' per the switch closed.  There may be other characters after the axis letter as grbl cycles through its check points each time the '?' command is sent.   

Note: for easier and quicker testing, set grbl to no hard limits ($21=0) to stop PicSender from issuing the hard limit alarm message and locking out controls.

cong-99, 2020-05-22:
Thanks - I'm afraid that this simply confirms that the switches are wired correctly.  I normally use PUTTY to talk with the controller.  I pressed the X, Y and Z limit switches, querying the controller each time in-between:  

<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:X>
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:Z>
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
ok

As can be seen, Pn:__ shows each switch activated and released.  I expect really that wiring switches and getting them working is something I shouldn't have a problem with, but it's a good thing to check! 

So now I've probed the X-driver signals (Step and Direction).  Here's the setup:
[image]

This is the resulting Step (yellow trace) and DIR signal when the X-motor is driven in the direction of the X-limit switch (using a G1 X-5 F200), i.e. working properly - you can see the yellow trace get denser over time, showing the X-step frequency going up (acceleration):
[image]

In contrast, this is the drive signal when a $H (home) is sent:  
[image]

You can see the STEP pulses - or lack of it - it's stuck in the HIGH state for some time. 

RESET then another $H attempt: 
[image]

As can be seen, it basically does the same thing. I am really stumped!! 

I hope someone can tell me where in the code to look: this is a planner error in GRBL or something... ??

By the way - one other thing - I am using the Arduino IDE to compile and upload the firmware (as suggested by the Github wiki page).  There are in fact a couple of compiler warnings:  

--------------------------

In file included from C:\Users\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Arduino\libraries\grbl/nuts_bolts.h:56:0: warning: "max" redefined

 #define max(a,b) (((a) > (b)) ? (a) : (b))

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files\Arduino168\hardware\arduino\avr\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

In file included from C:\Users\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Documents\Arduino\libraries\grbl/nuts_bolts.h:57:0: warning: "min" redefined

 #define min(a,b) (((a) < (b)) ? (a) : (b))

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files\Arduino168\hardware\arduino\avr\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

In file included from C:\Users\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Documents\Arduino\libraries\grbl/nuts_bolts.h:61:0: warning: "bit" redefined

 #define bit(n) (1 << n)

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files\Arduino168\hardware\arduino\avr\cores\arduino/Arduino.h:124:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

----------------------------------------------------------------------

All the above appear to be redefines of the same code, except for the last - I don't know if this is the source of the problem?  

Finally there's the STK() upload errors, which are addressed by choosing "Arduino Nano" and "old 328P bootloader" as the processor.

PicEngraver, 2020-05-22:
Well, you are working with a Woodpecker 3.2 board, and there is now a 3.4 available.  Perhaps you have a hardware, and not a firmware, problem?  Anyway, do you have an original Arduino Uno (not a China clone) you can test your compiled grbl with?   I do know there have been various reports of problems with grbl and Uno clones.  Might help to eliminate either the hardware or the firmware as the issue.  At any rate, I am out of suggestions myself.

cong-99, 2020-05-23:
Thanks for all the help - Found the problem!  

Boils down to "user stupidity".  I had the X, Y and Z max travel set to outlandishly large values (must have been thinking in micrometers).  The resulting Xmax must have caused an overflow somewhere (despite having soft-limits off).  

Conclusion - Woodpecker 3.2 Board is fine. Get the config parameters correct!!   

Closing this now... :-)

closed 2020-05-23 by cong-99