gnea/grbl #872: homing pull off issue

cornnatron, 2020-05-27:
Hello,

Just wanted to know if there is a way to reset this issue or to let it show correct
im running grbl to sv500 servodrives with servos and this where my settings

$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 2    (Step direction invert, mask)
$4 = 1    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.001    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 194    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 450.000    (X-axis travel resolution, step/mm)
$101 = 450.000    (Y-axis travel resolution, step/mm)
$102 = 2250.000    (Z-axis travel resolution, step/mm)
$110 = 2500.000    (X-axis maximum rate, mm/min)
$111 = 2500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 1000.000    (X-axis acceleration, mm/sec^2)
$121 = 1000.000    (Y-axis acceleration, mm/sec^2)
$122 = 1000.000    (Z-axis acceleration, mm/sec^2)
$130 = 600.000    (X-axis maximum travel, millimeters)
$131 = 300.000    (Y-axis maximum travel, millimeters)
$132 = 90.000    (Z-axis maximum travel, millimeters)

and everything ran fine on small scale but going to the limit of the machine,
i noticed a small error in distance so i started recalculating my steps close to the max distance the machine can do seeing that a small error would multiply over more distance.
correct me if im wrong please.

after recalculating these where my steps

 $100 = 444.264    (X-axis travel resolution, step/mm)
 $101 = 444.950    (Y-axis travel resolution, step/mm)
 $102 = 2222.500    (Z-axis travel resolution, step/mm)

only thing is now after a homing cycle it now pulls of -5.001 in y and -4.999 in x
z is still -5.000 before it was all -5.000.
any way to get that back to all -5.000?

terjeio, 2020-05-27:
> any way to get that back to all -5.000?

Short answer, no.
5mm * 444.264 steps/mm = 2221.32 steps -> 2221 steps sent to the stepper.
2221 /  444.264 = 4.999279... -> ~1µm error you cannot get rid of.

cornnatron, 2020-05-28:
thank you for the clarification .
but if i understand that right the numbers behind the decimal point are beeing droped or does that only happen when the sum of the steps isnt full steps ?
just trying to work out how it works.

terjeio, 2020-05-28:
There is no such thing as fractional steps. Rounding is to 3 digits behind the decimal point in metric mode and that may be increased by a simple code change. However with a step resolution of ~444 steps/mm the distance per step is 1mm/444steps -> just above 2 µm. Depending on the quality of your machine this is likely well below its inherent precision and thus the last digit does not carry any meaningful information. Precision/repeatability below 10 µm is for high end machines.

cornnatron, 2020-05-28:
thanks for that ill close the issue

closed 2020-05-28 by cornnatron

langwadt, 2020-05-28:
if you don't like the odd numbers you could rebuild the code with HOMING_FORCE_SET_ORIGIN then the position after pull off will be set to 0,0,0