gnea/grbl #878: rotate stepper motor

teobert97, 2020-06-17:
Hello everyone, there is a command in the GRBL library that allows me to rotate an axis (stepper motor) at a certain speed almost indefinitely, that is, without defining the length I want to reach but only the speed at which I want it to run.
Thanks !

bdurbrow, 2020-06-21:
I presume you are asking if such a function exists.

No, it doesn't, and adding it would be difficult on the '328 platform, as there just isn't any flash space left available.

On the Mega 2560 platform, it could be done but it would be quite a bit of work; you'd have to rework both the stepper ISR and the planner to make it function properly.

Why do you need this? Are you trying to drive a non-positioning motor? If so, there are several other ways to go about this; either using the spindle rate output, or sending a command to an outboard processor (an Arduino Nano or similar would do; or if you are doing your own PCB, you could add something like an ATTiny, just to drive the motor).

teobert97, 2020-06-22:
Thank you so much for such an exhaustive answer!
Yes, I am trying to make my own pcb to control the stepper motor without any position, only with the rotation speed. Can you explain to me how you would do in this case? with some schematic and parts of code.