gnea/grbl #882: M8 command - which pin, what signal?

cvhorie, 2020-06-24:
I am using (or at least moving in that direction) Arduino/GAUPS(a gShield clone)/GRBL/Pascal to move an Olympus camera over a surface.
Once in position, I need to tell the camera first to focus then to activate the shutter. Each is achieved by shorting the appropriate camera control wire to GND.

As a novice, it seems that M7 and M8 are the most appropriate commands to use.  
I should be grateful to be pointed to the file that specifies the appropriate pins and the signal GRBL sends to them.  I can then use relays to short the wires.
More generally, it would be useful to have this information for other commands.

Any advice gratefully received.
Thank you.

langwadt, 2020-06-24:
https://github.com/gnea/grbl/blob/master/grbl/cpu_map.h

cvhorie, 2020-06-24:
Thank you. I would have struggled to find the file, if I had know it existed.
One remaining piece of useful information is: 
If I call M8, what happens to the signal on Analogue pin 4?  Is it an analogue 0~5 V, or a digital 0>5 V?

langwadt, 2020-06-25:
it is a digital on/off 0/5V 

https://github.com/gnea/grbl/blob/master/grbl/coolant_control.c

cvhorie, 2020-06-26:
Thank you. I guessed the wrong way round.
At my stage of ignorance, I need a cook book - but I do like to know where the ingredients come fom.
You referenced Dr Jeon's file (and what a lot of work he has put into GRBL).  It does not mention the physical output you kindly gave.  Is there a (simple?) description of the electrical outputs which are generated by the programmed commands, or is this board dependent, or too interdependent?