gnea/grbl #883: M30

entolium, 2020-06-26:
Hello!

I send this program to my GRBL 1.1:

G91
G01 X150 F300
M30

Then, GRBL stops to send me the status report while a few seconds and raise the ALARM:3 (Reset while motion).
After the alarm raise, the X position is less than 150 (117.6).

I think GRBL execute the M30 code when is received, but should wait to finish the previous command (G01 X150...)

Thank you and sorry for my english...

neilferreri, 2020-06-26:
I think you have something else going on.
What happens when you don't send the M30?

entolium, 2020-06-26:
without M30, the commands are executed without prlobem

neilferreri, 2020-06-26:
What sender?

entolium, 2020-06-26:
DIY
In this code example, UGS finish the program (reading the verbose output) but don't send the last block (M30).
Could be because they found the same problem with M30 and finish the program manually.

neilferreri, 2020-06-27:
The M30 won't "interrupt" your movement.  Grbl will not move on to the next command while still sending step signals.  Is the distance you moved the same as that reported by the controller?  Send a `?` after your M30...what position does it report?

entolium, 2020-06-27:
I can send thousand of blocks without problem, and grbl runs perfectly.
I read continuously the status report and send the number of free blocks (reading the buffer free blocks).

But, if I send the M30 code, grbl stops immediately to respond to the "?" command and after a few seconds raise the alarm3 and stops the movement. This is the problem and I can't find an explanation.

For example:
Start position: [X0 Y0 Z0]
G01 X50 F500
G01 Y50
M30
Alarm3 and End position: [X38.125 Y0 Z0]

I'm testing with an Arduino Mega connected via USB to my laptop (not in my CNC), and I think this is not a noise problem.
$100=40.0
$101=40.0

I'm thinking if the problem could be the Mega version of GRBL. I will try with Arduino UNO.

entolium, 2020-06-27:
I tested with Arduino UNO (GRBL 1.1e) and the same problem

neilferreri, 2020-06-28:
Try the most recent version.  I cannot replicate what you're seeing on 1.1f or 1.1h.

entolium, 2020-06-28:
Hello!

I found the problem.

When you send "M30", grbl don't respond with "ok", and my program raise an error of timeout and send a reset. I'm changed my code and now runs ok.

Grbl should respond "ok" to "M30" code to keep a coherent behavior.

Thank you very much for your help and patience.

closed 2020-07-03 by entolium