gnea/grbl #885: List of compatible boards missing?

maxholgasson, 2020-07-01:
Is there a list available with compatible boards? I'm interested if this board is compatible with  the latest grbl version?
https://www.amazon.de/QPX-Triaxiale-Steuerplatine-Controller-Bildschirm/dp/B07D211NKD/

paulvdhoeven, 2020-07-08:
I have cobbled something together with a Blue Pill and some 74LS244 TTL driver chips on a piece of Vero-board. The TTL chips drive the opto couplers  from external stepper motor drivers.

Boards such as in your link are usually pre-loaded with some variant of CNC firmware. (In this case GRBL)
Personally I would never buy such a board if it does not:
A). Specify which processor is used.
B). Does not have a link to source code.

AtMega328 is also too near it's limits and cracking at it's seams for GRBL. There are multiple on-going efforts for ports to more capable microcontrollers, and my closest guess is that AtMega328 is "Legacy".

"Bigtreetech" may be a nice company to deal with, and still is quite affordable.
It is maybe not the cheapest clone you can find, but good (Does BigTreeTech do 4 layer?) PCB's are worth something too, and making documentation available also costs time, and thus money for a commercial company.

Here is a link to one of such boards from Bigtreetech:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4 

For a more general overview of GRBL compatibility with different processors, have a look at:
https://github.com/terjeio/grblHAL/

andyb749, 2020-10-04:
> Is there a list available with compatible boards? I'm interested if this board is compatible with the latest grbl version?
> https://www.amazon.de/QPX-Triaxiale-Steuerplatine-Controller-Bildschirm/dp/B07D211NKD/

I've got one of these QPX boards.  Bought it a few weeks ago and came flashed with the version 1.1h.  

Output from $I is: 
```
$I < [VER:1.1h.20190825:CUSTOM]
[OPT:VL,15,128]
ok
```