gnea/grbl #886: work position of y axis starts at 10mm when connected

Kruze17, 2020-07-03:
Greetings and thank you for taking time to read this problem i am encountering.
Whenever i connect my grbl loaded(1.1) arduino nano to UGS it connects but the work position is always at positive 10mm.
I tried to find what is making this happen but no success. Any kind of help, tip or suggestion is appreciated.
I have attached a capture of the UGS.

Thanks in advance.
[image]

langwadt, 2020-07-03:
work offsets are persistent, try reset y axis

Kruze17, 2020-07-04:
> work offsets are persistent, try reset y axis

Thanks for the suggestion... Resetting the y axis works for me.

closed 2020-07-04 by Kruze17