gnea/grbl #887: GRBL 1.1: Simulation of NC file is correct, but real toolpath is crazy

Chris-the-electrician, 2020-07-03:
Hi everyone,
I've experience a problem with Arduino UNO and GRBL v1.1e, currently milling/engraving a motorcycle's logo.
This logo has a pocketed frame around, which is partly curved and partly straight. (actually, it's the H-D emblem).
The milling simulation looks 100% OK, but in reality, approx. after engraving half of that pocket, the curved part of the pocket (lower left area of workpiece) becomes stretched in X-axis (tool paths split to then end of the curve, but very smooth?!). At the end of the job, the left vertical part of the frame has twice of the expected width (half pocket has an offset about 2 mm). But on the left half of the workpiece, everything is still OK! (the stretching effect on the left side is compensated again on the toolpath to the right?!
It can't be lost steps, because the toolpath is extended in one direction (-X), and in opposit direction (+X), the toolpath is shortened, and the most right point has no offset.
The upper left curved part starts with split paths, and when the curve ends, the toolpaths are no more splitted!
Is GRBL making rounding errors, calculation errors, having timing problems?
Steppers sound very smooth, axes moving easy (not blocking), no backlash (pre-loaded ball screws). Moderate feed rates (tried different speeds, nothing changes).

Has anyone experienced something similar?

Regards,
Chris

MeJasonT, 2020-07-05:
Hi Chris,
treat yourself to Camotics
[https://camotics.org/](url)
Its an open source visual simulator, really good for proving your tool paths befor commiting

your problem sounds like your steps/mm are out or you may have some backlash in your machine.
have you calibrated your x axis recently, ie you move it 100mm and its pretty much 100mm return to zero and it goes back to exactly the same point.  If you set up your axis over a short distance say 10mm and there is a error of 0.01mm then when the scale of your part is 10x larger so is the error - so in this example would become 0.1mm.  Its a pain in the bum and takes ages to get it just right but i would wager thats where your error is.  The good news is it sounds like you have narrowed it down to your X axis and not an issue with both.  As the Y axis on machines is shorter in travel it tends to be the one that shows calibration errors the most.  There is a formula to help get steps/mm corrected
new_steps = old_steps * (100 / distance_actually_moved) this should get you fairly close then you will need to experiment and tweek.  I find milling a circle profile or pocket into a piece of scrap helps with checking the accuracy of your machine.  Its checking distances and arcs both at the same time. make a shallow cut and measure against what you expect.  Increase your cuts depth to reuse the scrap for the next test, an 8mm or 10mm thick block should give you a few good goes at testing before you have to ditch the material and set up the workpiece again.  You should be able to see any step difference visually straight away.  

if you get a perfect cut around the same shape every time then either nothing in firmware is changing or your in deep doodoo cause its doing things no one could explain. lol.

Chris-the-electrician, 2020-07-06:
Hi Jason,
as I've written in my starting post, in the simulation it looks good. I'm using CamBam 0.9.8 and the Camotics sim. The toolpaths look perfect on the whole workpiece.
On real workpieces, a sqare is a sqare, a circle is a perfect circle, an the workpiece dimensions are 100& accurate!
Just this fu**ing piece of logo with this "concave rounded corners" is stretched (on -X end only!).

MeJasonT, 2020-07-06:
Hi Chris,
i can see whay you are saying.
i dont think your issue is software related as the image would turn out garbage.  over and undercuts usually point towards the steps/mm or backlash issues.  I have an issue on my machine when it gets to the end of +X travel which is down to the lack of wear on the ways (its an old mill and its done most of its work at one end of the table) preloading the x ballscrew on my machine is a bit of a science.  without knowing how you set your work piece up, i can only guess the you used the top ight corner as we look at it as the datum (the centre line would be somewhat of a mis direction) the whole image would be elongated on one side and tapering towards the other in the top right WCS datum (0,0,0)   from what it looks like both the X and Y axis are running wide at the far end of the workpiece.  M on the motor is half the stepover agin (X axis) as is the horixontal below HAR (Y axis).  You have definately trammed your bed at its not miles out.  Put a felttip/marker pen into your chuch and draw a line along one of your t slots from one end of the bed to the other and just make sure your not running off on a tangent do the same on Y also check y line is square  against the side of the bed.  It does smell like somthing mechanical.  If the circular intropolation the I & J moves to you and me would cock up on any angle or circular shaped letter if GRBL was to blame so it has to be X and Y moves.  Whats it like size wise overall, if you drew a square around the cad and then using the parrallels at H & N letters and the height from oto and cl at the top and bottom is the workpiece similar in dimension to the cad.  better still print the cad on acertate of greaseproof paper (somthing see through) and see what that throws up.  you already know your machine is running wide at one end,  Its like your bed is making a circular movement  +x is going too much +/- in y direction and then when it returns back to -x the error is very small so the image is near perfect.  Im suprised the circle test didnt throw anything up, judging by whats here it should have made an egg.  Try another design of similar size and see if it shows the elongation on that aswell.  There is no suggestion that you are missing steps however the amount of steps you need to travel a certain distance can be inacurate which would show this kind od error. if your stepper is typically 200 steps and your Ball screw is 1 revolution =5mm the stepper driver is set for 1/8th step for example then you would require 1600 steps to move 5mm doe to the mechanics of your machine 1600 steps might come out at 4.998mm or 5.010 that sort of inacuracy is what gets amplified into your work piece.  oddly one end of your design is correct whilst the other is miles out, reason is it knows exactly where 0,0 is but 50mm along your x suddenly becomes 50.1 if the steps mm are out (5.010).  I have used a dial guage and bounced backwards and forwards to get the steps/mm bang on only to find 200mm along the axis the measurements were out.  the pro manufacturers use a ballbar to calibrate machines for this very reason.

Chris-the-electrician, 2020-07-06:
Hi Jason,
the machine is like brand-new, it has only worked about 20 hours until now. All axes are equipped with ball screws 20x4mm, and are absolutely backlash-free (expensive pre-loaded stuff).
I built the machine in co-operation with two students of a technical high school in their diploma year. The machine is built really tough and rigid, (frame made of aluminium profiles 60x60mm, some of the profiles are doubled (in parallel), to make it even more rigid.
All axes and lead screws are adjusted with calipers and gauges (0.01mm), there is no wear-out or backlash anywhere.
The workpiece-origin is in its center (see camotics screenshot), which is also approx. in the center of X- and Y-axis.
-> The funny thing is, that the strange offset only happens in the left half of the part, and also, that the offset starts after the half of the toolpath in this area (e.g. the complete pocket toolpath consists of 10 rounds from inside to outside, round #1 to #5 is OK, and in round #6 it starts to mill in offsets!). Have a look at the narrow island stripe on the left half of the pocket (red dotted line).
I've tried this H-D logo with different feedrates (from 250 to 350 mm/min), to check for unwanted vibrations (resonance frequencies of stepper motors), different mill diameters from 0.2mm to 0.8mm (the absolute dimensions / pockes sizes were not the problem, the problem is the basic geometry offset / distortion between left and right side.
And for milling, currently I use the GRBL machine plugin.
In CamBam, the generated toolpaths look also good (see picture).

Comparing all three pictures, it looks like the outer "frame" (the shape with the red and green marking on the real workpiece)
starts to have an offset after half of the "rounds".

[image]

MeJasonT, 2020-07-06:
My assistance has exceeded my intelect unfortunately. Im baffeled as to why it should be fine on one half and not the other.
There could be a negative number rounding issue or positive integer roll over in software but im sure there would be loads of people shouting about it long before now.

there are two other options you can try
Camotics has proven the file so the cam is spot on.  You trust the machine 100% so the only other variable is the interpreter/GUI (cambam plugin).

Try using a different gui/interpreter like GRBL_Panel, Candle or UGS, it may be an issue with the control side of Cambam - have you raised a question on their forum.

Secondly, move your wcs from centre and run the job from a corner, if its getting pissed running calculations from centre it will have to figure things out running from one side instead. Sneaky. 

I would have expected the reverse of the problem to be true for any negative coordinate rounding issues when the right hand side is effectively in negative space from the datum.  Hopefully somone else will chip in who has the magic answer.

Chris-the-electrician, 2020-07-07:
Hi Jason,
could it be, that the GRBL machine plugin sends some lines twice or multiple times?
With my recent parts, I never had problems like this. In the past, I milled parts for RC planes, levers for RC servos, also I've milled a perforated disc with 60 slots/holes for a tachometer (rpm counter). Everything was perfect with Cambam and GRBL machine plugin.
That's why I am a little bit concerned about a GRBL firmware/algorithm fault...
But I will try to mill the H-D logo with different software like UGS, or GRBL controller, or CNCJS.
Could take a couple days for that...

Thanks for your effort!

Regards,
Chris

paulvdhoeven, 2020-07-08:
I can not combine what I see in your pictures with something that could be related to Grbl not working properly.
You say it's a brand new self designed machine.
How much experience do you have with engraving in Aluminimum?

I see a lot of smearing, and chips sticking to the work piece. Especially for example in the inside of the "A" of "HARLEY".
It looks like you have not used any lubricant.
Usually Aluminimum uses some liquid as lubrication and anti-stick. Aluminimum is a gummy material and it tends to stick to the endmill, and from there it gets smeared all over your work piece. To prevent this usually a mist spray or (or some drip system) is used to prevent this. For Aluminimum usually some light oil or even WD40 tends to work well. Some people use alcohol (Regular Ethanol, or Isopropanol) This has less anti-stick properties, but has the advantage that it evaporates and no de-greasing or cleaning is needed.

This smearing can also be caused by a blunt endmill (too high RPM, can dull a mill quick), or the wrong endmill altogether (Different mills for different materials, form, coatings, etc).

Other possible problems are too long tool stickout, and climb versus conventional milling. Even if your CNC machine has no backlash, Climb/conventional milling matters because your tool will bend because of the side-loading, and this can not always be neglected.

You seem to have used a ball-nose mill, and this is not optimal. The cutting speed of the center of the ball is always 0. You'll also get nicer results if you use 2 tools. One ball or slanted for the sides, and one with a flat bottom for the bottom.

If you want to verify whether the problem is with your tooling, lubrication, etc, or with tool path generation, then put a simple (spring loaded) ballpoint or felt tip marker in your machine and draw the thing on paper. For my own CNC, I've made a bracket from an old HDD magnet, a springy piece of aluminum and some wood sticks, and a filling of a Parker pen, which is permanently hot glued to the contraption. It has a few mm of springiness to get some preload on the pen, and it is very quick to mount with the magnet. The magnet also simply shifts or gets torn loose on gross programming errors without any permanent damage.

Chris-the-electrician, 2020-07-09:
Hi Paul,
Yes, I've used a ball nose engraving tool (dia 0.8mm x 36°). I know, that a flat end mill would be much better for the pocket ground surface, and I know that the center speed is zero.
I use air/mist cooling on my machine (with high performance cutting oil, similar to WD40, because alcohol vaporises very fast and you get dizzy, it stinks a lot), this worked very good on recent parts, made of the same material, some of them with the same cutter bit.
Machine is running on an industrial 24v power supply. It's also equipped with a high frequency spindle (0.8kW) and a proper frequency converter (Sinamics G120).
All the frames of my machine are made of aluminium profiles, 60 x 60mm, some of them are doubled.
I adjusted the whole machine, using dial gauges (0.01mm resolution) and reference blocks (granite).
If I try to push ANY of the axes back and forth, or the end of the spindle/tool by hand (force is about 300 - 400 Newtons), the gauge stays within +/- 0.02mm. Even if I push harder, th gauge barely moves more than this.
The drives are NEMA23 motors, with ball screw drives (12x4mm), backlash-free (the expensive stuff, not the chinese!).
I measured the linear force of all axes in lower stepper rpms (120-150rpms), the testing apparatus with load cells (our company is a manufacturer of industrial machinery) showed a force of almost 2000 Newtons (= 200 kilograms). -> That should be enough to avoid loss of steps. This speed is about the max feed rate I am milling / cutting off material). Usually, I set the feed rate lower than that (250 -350mm/min) to get a smoother surface. Rapid movements are limited to 2500mm/min
This time, I tried different feed rates, makes no difference in the toolpath (no bad resonance frequencies). Tried different spindle speed (12k - 24k), no difference in the toolpath.
There are no shaky edges on the pockets, the machine is rigid enough.
I just looked again at the real workpiece above, and I realized, that the top of the outline of the H-D logo (the "roof" of the logo) has a slight offset to the left (-X).
I also generated seperate NC files for the text and the outer "frame", the problem is always on the same place of the frame. This frame gets an offset anywhere in the middle of the job, like 5 rounds of toolpath are ok, and then GRBL/Arduino makes an "offset jump", before it continues milling.
I generated the gcode ONLY for the outer pocket, checked it in Camotics sim, ran the job, and, half way done, the toolpath had the same offset. No matter, which feed rate (250/300/350mm/min), or spindle speed...
After this, I've made milling tests with squares and circles (even with higher feed rates and 2mm deep in the material!); checked the dimensions of this parts - maximum deviation was +/- 0.02mm! Very good.
In the moment, I have no clue whats going on.
Anyways, I will examine the NC file closely, and compare it with the toolpaths in Camotics, and in real; to find out the reason for this problem. I also will try different control software, like UGS, or CNCJS. Let's see, what happens...

Regards,
Chris

Schildkroet, 2020-07-09:
Can you share a small piece of code which causes the problem? Maybe i have some time to test it on my machine.

paulvdhoeven, 2020-07-09:
You seem to know what you're doing, it's always hard to estimate that from a bit of text and an occasional picture.
I also can not see in which order the thing was milled. and did not hear the machine, etc.

The smearing and stuck chips do look like a tool problem (blunt or lubrication), and it may be a separate problem from the lost steps in X direction. Carefully check the wiring of the X-axis motor. Loose wires don't pass through pulses :)

Another problem that sometimes arises is that the optocoupler for changing direction often is slower (cheaper) than the optocoupler for the step pulses. and a small delay when changing direction may be needed. I checked this with my own CNC machine and GRBL and it was OK, that was a few years ago though. It also depends on the wiring of the stepper motors, for example if you trigger a step on the positive or on the negative edge.

Drawing the thing on a piece of paper is very cheap to do, and when you wiggle the wiring while it's drawing on paper, problems are likely to become obvious.

Loss of steps can also easily be measured if you move the machine to a reference position after the job, and then measure it's position. If loss of steps is because of mechanical overload then the deviation will always be a multiple of 4 full steps (for a 2 phase motor)

Yet another possible cause of problems is your arduino uno board. The thing was developed by a few students some 15+? years ago, and one of the mistakes they made was to use a 16MHz crystal, which does not play nice with baudrate generation, and this can eat up about all your timing margin on the serial port, and generate occasional errors. Long time ago I read some lines about this on https://github.com/vlachoudis/bCNC/wiki/ The cheap clones seem to suffer most from this (ceramic oscillator with bigger tolerance) but also the "original" arduiono boards are not immune to this.This is one of the main reasons I never used an arduino board. Instead I used an STM32f103C8T6 (aka "Blue Pill") which pushes the data over USB, and USB has checksums (and I think retransmission) built into the protocol, and at least does not suffer from the timing problems of the Uno.

The milled out parts are very rough... Which is a good adhesion base for paint :) You can fill in the low areas with paint.

Chris-the-electrician, 2020-07-10:
Hi guys,
today I tried to engrave this part on a piece of plastic (less cutting forces), and the offset problem got even worse! I checked the axes drives for backlash etc. and found out, that on the X-axis the shaft coupling had a lose screw. Shame on me! During acceleration and deceleration, due to inertia, the ball screw was moving back and forth, causing this offset. There is an adaptor sleeve for  a smaller shaft diameter inside the coupling. This sleeve (a piece of aluminium tube with slots) had no grip on the drive shaft. I re-tightened the coupling, and it works like a charm. Following testrun was performed with higher a feed rate.
Thanks a lot for your efforts!

[image]

MeJasonT, 2020-07-10:
See i told ya.
Chris stick to electrics - lol.

MeJasonT, 2020-07-10:
Pleased its fixed buddy.

Chris-the-electrician, 2020-07-11:
Hey Jason, you cheeky guy :)
That little machine is working again, and I'm verry happy about that.
I found some pictures of my machine, made during it's construction and assembing.
Some details about the project:
The CNC was built in cooperation with two guys of a technical high school, as their graduation project.
As they had no experience in real mechanical construction, and had no clue about CNC machines anyways, it was pretty hard at the beginning. At school, they mostly did theoretical calculations, but they had no idea about the real hardware stuff. On the first design, they wanted to assemble the whole machine frame with M4 screws, because the calculations said, it's strong enough. But this dimension is for kiddy toys. I also told them, to set some braces here and there, to get a better rigidity, and so on...
Another important point of the project was the evaluation of hazards, because the machine will not be used in a private hobby workshop, but in our company's workshop. So there are some (a lot of) rules to follow: emergency stop, interlocked protective doors, limited speed for movements when door is opened, and so on.
Also we had to write an operator manual, I made the electrical schematics (I am doing electrical engineering in our company).
Some of the mechanical parts were handmade as well, they don't look perfect, but they work.
I made the control cabinet by myself, I did not want to make it perfect, because it should look like made by the graduates.
And some of the PCBs in the control cabinet were self-made; quick and dirty, but it works :-)
:-)
[image]
[image]
[image]
[image]
[image]
[image]
And, finally, the machine got it's own place in our workshop:
[image]
[image]

closed 2020-07-11 by Chris-the-electrician

MeJasonT, 2020-07-11:
Hi Chris,
That is a very nice build, really like the panel, top quality.  My machine is undergoing a transformation, sadly im leaving GRBL, not entirely as its still going to remain on the plasma table.  The mill is now a PathPilot clone.  I found GRBL too limiting with its limited Gcode command set, not that it isnt total fantastic but unfortunately i grew out of its functionality.  It would be like owning a Denard and upscaling to a Haas.  Im not ruling a haas out either so the pathpilot control will probably go eventually.
My mill is a converted Knee mill, I also wired my own cabinet and built my own PCBs.  To think i started off 8 years ago with a 3D printer after watching Jon Saunders NYCCNC on youtube.  Sadly the quad copter i had the ambition to build never took off so to speak, just aswell it was just under 2 kg frame built from an aluminium Tig welded frame.  Oh the things true engineers have to be capable of..  To hear the graduates were fine of the paper calculations didnt suprise me,  I have a voulenteer apprentice (he does not get paid).  He finished at university in Glasgow UK and throughout the whole of his engineering degree was not allowed to use any machines, very little practical included in the course.  He was mad enough to want to come and learn from me.  took about 10 minuets.  As far as snowflakes go he is an absolute star, he has the drive and ambition i had at his age, I cant say its a quality i have found in many others of his age.

One day i will finish my machines enclosure, when its not busy.

pm me if you want to stay in touch
J.

[image]
[image]
[image]